ControlSystemStudio/phoebus #1082: medm converter issues

MarkRivers, 2020-01-26:
I seem to have found 2 issues with the medm converter:

- If the output file already exists I get an error.  Is there a way to force it to overwrite the output file if it exists?

- It seems like the command to do the conversion returns before the conversion is actually complete.  This makes it difficult to use with "make" because actions to be taken after the conversion command won't work because the output file does not yet exist.

Note below that the new command prompt had already appeared before the output from the converted started.
```

corvette:measComp/measCompApp/op>make -sj4
corvette:measComp/measCompApp/op>2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/measCompWaveGenPlot.adl -> bob/autoconvert/measCompWaveGenPlot.bob
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/1608G_2A0_module.adl -> bob/autoconvert/1608G_2A0_module.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/1608G_V5.adl -> bob/autoconvert/1608G_V5.bob
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/1608G_V3.adl -> bob/autoconvert/1608G_V3.bob
2020-01-26 12:39:55 CONFIG [org.phoebus.product.Launcher] Loading settings from /usr/local/phoebus/phoebus-4.6.0/settings.ini
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/measCompAoSetup.adl -> bob/autoconvert/measCompAoSetup.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/measComp_pulse_generator.adl -> bob/autoconvert/measComp_pulse_generator.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/C9513_module.adl -> bob/autoconvert/C9513_module.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/1608G_V4.adl -> bob/autoconvert/1608G_V4.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/measCompDemoTop.adl -> bob/autoconvert/measCompDemoTop.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/measCompADCStripChart.adl -> bob/autoconvert/measCompADCStripChart.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/C9513_chip.adl -> bob/autoconvert/C9513_chip.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/C9513_counter.adl -> bob/autoconvert/C9513_counter.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/measCompWaveDigPlot.adl -> bob/autoconvert/measCompWaveDigPlot.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/TC32_module.adl -> bob/autoconvert/TC32_module.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/USB231_module.adl -> bob/autoconvert/USB231_module.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/1608G_module.adl -> bob/autoconvert/1608G_module.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/USBCTR_MCS_plot.adl -> bob/autoconvert/USBCTR_MCS_plot.bob
2020-01-26 12:39:56 INFO [org.csstudio.display.converter.medm] Convert adl/USB2408_module.adl -> bob/autoconvert/USB2408_module.bob
2020-01-26 12:39:57 INFO [org.csstudio.display.converter.medm] Convert adl/1608G_V2.adl -> bob/autoconvert/1608G_V2.bob
2020-01-26 12:39:57 INFO [org.csstudio.display.converter.medm] Convert adl/1608G_V1.adl -> bob/autoconvert/1608G_V1.bob
2020-01-26 12:39:57 INFO [org.csstudio.display.converter.medm] Convert adl/measCompTop.adl -> bob/autoconvert/measCompTop.bob
2020-01-26 12:39:57 INFO [org.csstudio.display.converter.medm] Convert adl/USBCTR_MCS_plots.adl -> bob/autoconvert/USBCTR_MCS_plots.bob
2020-01-26 12:39:57 INFO [org.csstudio.display.converter.medm] Convert adl/E1608_module.adl -> bob/autoconvert/E1608_module.bob
2020-01-26 12:39:57 INFO [org.csstudio.display.converter.medm] Convert adl/ETC_module.adl -> bob/autoconvert/ETC_module.bob
2020-01-26 12:39:57 INFO [org.csstudio.display.converter.medm] Convert adl/USBCTR_MCS.adl -> bob/autoconvert/USBCTR_MCS.bob
2020-01-26 12:39:57 INFO [org.csstudio.display.converter.medm] Convert adl/measCompAiSetup.adl -> bob/autoconvert/measCompAiSetup.bob
2020-01-26 12:39:57 INFO [org.csstudio.display.converter.medm] Convert adl/USB1208LS_module.adl -> bob/autoconvert/USB1208LS_module.bob
2020-01-26 12:39:57 INFO [org.csstudio.display.converter.medm] Convert adl/measCompTemperatureSetup.adl -> bob/autoconvert/measCompTemperatureSetup.bob
2020-01-26 12:39:57 INFO [org.csstudio.display.converter.medm] Convert adl/USBCTR.adl -> bob/autoconvert/USBCTR.bob
```

kasemir, 2020-01-27:
The Linux launcher script (phoebus.sh) is what created a background process. When called with `-main ...` to run a converter instead of the default UI, it'll now run in the shell process.

Also added a `-force` option to replace existing output files.

Will look at the message button <=> bool button convertion, https://github.com/areaDetector/ADCore/issues/411, and the put that all into one pull request.

closed 2020-01-28 by kasemir

kasemir, 2020-01-28:
Mark, you can get the updated binary from 
https://controlssoftware.sns.ornl.gov/css_phoebus/nightly (phoebus-linux.zip, ..)
which includes the -force option etc.

MarkRivers, 2020-01-28:
I have tested the new version and it fixes both problems.  I have updated RULES_OPI to use the -force flag.

Thanks for the quick fixes.

PR #1083 (2020-01-28): MEDM converter
Fixes #1082 by adding a '-force' option to the MEDM converter and having the launcher script run it in the invoking shell process.

Fixes 'button' version issue mentioned in https://github.com/areaDetector/ADCore/issues/411 and while at it tried to make most of the 'synapps' displays convert without fatal errors.

Remaining issues that won't be fixed:
 * No support for 'octal' number format -> Shows decimal.
 * Macro 'args' in MEDM file must be comma-separated as also shown in https://epics.anl.gov/EpicsDocumentation/ExtensionsManuals/MEDM/MEDM.html MEDM manual. Not supporting s [... cut]