ControlSystemStudio/phoebus #1088: Widget properties editable despite use of class

georgweiss, 2020-01-30:
Reproduce as follows:
1) Create a new display, which will load the "template" where for instance the label using the TITLE class will disallow user to modify font and other properties.
2) Shut down Phoebus and re-launch.
3) In the new display the properties of the label previously locked are now editable.
4) Close the tab with the display tab and re-open in edit mode from file browser.
5) The properties of the label locked according to TITLE class.

kasemir, 2020-01-30:
Can't reproduce.  After 2. Shut down Phoebus and re-launch, 3. the TITLE label properties in the automatically re-opened editor are still locked.

When you get into the state 3. where the properties are editable, will the editor context menu "Re-load Widget Classes" put the editor back into the correct state? Or "Re-load Display"?

kasemir, 2020-01-30:
Could be a timing issue, see `WidgetClassesService`. When we open an editor, we don't want to block everything until the class support has been loaded, since that could take a long time in case your *.bcf file is fetched via NFS or HTTP.

When you restart Phoebus, do you see the warning "Cannot use widget classes, still loading"?

kasemir, 2020-01-30:
The timeout is configured via https://github.com/shroffk/phoebus/blob/master/app/display/model/src/main/resources/display_model_preferences.properties#L39
```
# Timeout [ms] for loading files: Displays, but also color, font, widget class files
org.csstudio.display.builder.model/read_timeout=10000
```

georgweiss, 2020-01-30:
No warning. Cannot see why a 10 second timeout should not be sufficient on my test systems.
Anyway, I reproduce on Mac, Linux and Windows. If the label in question is selected and I reload classes, nothing happens in the properties panel until I select some other component and then reselect the label. Reloading display does the same thing, only the label selection goes away as a consequence of the reload operation.

kasemir, 2020-01-30:
Are you setting `org.csstudio.display.builder.model/class_files` to something other than `examples:classes.bcf`?

In the editor, does the TITLE label position and font look as if it's handled by the class, yet it's still editable? That would be an editor issue.
Or does it look like it doesn't know about the TITLE class? That would be a *.bcf class file loader issue.

The way it's supposed to work:
1) The display is loaded via `ModelLoader.loadModel`
2) It calls `WidgetClassesService.getWidgetClasses().apply(model)`
3) When `WidgetClassesService.getWidgetClasses()` is called for the first time, it waits a little for the class file loader, or times out.

If you can reproduce the issue, then try to check what's happening in that chain.
Like add a printout in `WidgetClassesService.getWidgetClasses` to check if it really loaded something, or maybe didn't but for some unknown reason there's no error message.

georgweiss, 2020-01-30:
This is is all with default initial.bob and classes.bcf.
First time getWidgetClasses is called (when the editor GUI is created I guess) the classes have not been loaded. Next, loadWidgetClasses is called as a consequence of calling ModelLoader.loadModel.
Maybe this is a race condition. There's a number of things triggered on different threads.

georgweiss, 2020-01-30:
The style of the label seems to be taken from the TITLE class, even when they are editable.

georgweiss, 2020-01-30:
By the way, first time getWidgetClasses is called the Future object is null and the method returns a WidgetClassSupport object where no classes have been loaded. It waits only if loadWidgetClasses has been called first.

kasemir, 2020-01-30:
WidgetClassesService.getWidgetClasses() waits for the loading of widget classes IF that has been started.
It's initiated by calling WidgetClassesService.loadWidgetClasses() from ModelPlugin.reloadConfigurationFiles().
That in turn is a static initialization, for example triggered by accessing the logger, and that's the unpredicatable race condition.

As a test, try to force initialization in the editor app by adding this to DisplayEditorApplication`:
```
    @Override
    public void start()
    {
        ModelPlugin.reloadConfigurationFiles();
    }
```

kasemir, 2020-01-30:
Probably better to put that `start()` code into the `DisplayRuntimeApplication`, not the `DisplayEditorApplication`

kasemir, 2020-01-30:
yes, that does it. I'll have a PR soon,

georgweiss, 2020-01-30:
Will try tomorrow.
By the way: initial.bob defines the TITLE class, as does classes.bcf. Which definition is actually used?

kasemir, 2020-01-30:
initial.bob USES, classes.bcf DEFINES.

Initial.bob says: I'd like to use TITLE, but in case that's not defined, use the following settings.
Those just happen to match what classes.bcf defines.

kasemir, 2020-01-30:
When a widget uses a class, it still saves the font etc., all those things set by the class.
That way, if you don't have that class defined, you still see what the person who used the class saw.
If you then define the class, your class settings are used.

kasemir, 2020-01-30:
So the initial.bob shows that the one label was created with the TITLE class, and at the time that meant x, y, font, ... were set by the class as shown in the file.

If you open that initial.bob without any classes defined, you get those last class-provided settings. The idea was that creating a file at site one with some class settings, then emailing that file to another site, it should still "work", not suddenly having everything undefined. We want to avoid the issue where you get a *.html file without *.css, and suddenly it looks all rubbish. Our displays look as they last appeared when edited, but you're free to apply your own class which will override anything in the saved file.

If you open that initial.bob with a TITLE class defined, whatever you have in that TITLE class applies. The "use_class" attributes in the initial.bob are just a hint that those settings were set from a class when the file was edited the last time. It doesn't re-define the class. Classes are only defined in *.bcf files.

kasemir, 2020-01-30:
https://controlssoftware.sns.ornl.gov/training/2019_USPAS/Presentations/05%20DisplayBuilder.pdf pages 48/49 try to explain.

georgweiss, 2020-01-31:
Your PR seems to do the trick, thank you.

closed 2020-01-31 by georgweiss