ControlSystemStudio/phoebus #1092: Application menu bar unresponsive on Mac OS 10.15+

georgweiss, 2020-01-31:
The application menu bar is unresponsive on Mac OS 10.15+ after launch of Phoebus. User needs to select another app and then switch back to Phoebus to be able to use the menu.

This seems to be a JavaFX thing and is reported here: https://bugs.openjdk.java.net/browse/JDK-8233678. Tried EA releases of JavaFX 14 and 15, but that did not change the behaviour, even though the bug report states that a fix should be available in v 14.

Will monitor this as new releases of JavaFX 14 become available.

kasemir, 2020-02-17:
Thomas Birke found https://stackoverflow.com/questions/58624764/javafx-global-menu-of-javafx-applications-unresponsive-on-macos-10-15-catalina

It suggests that the 'main' class and the 'Application' class need to be separate  -> Already the case with our `Launcher` vs. `PhoebusApplication`.

It further suggests to call some `Desktop` code before `Application.launch`, which we could try.

georgweiss, 2020-02-17:
Already tried the suggested solution without success.

kasemir, 2020-03-24:
For what it's worth, the menu bar has been OK for me with JavaFX14 #1230. It populates within about a second after the 'splash' closes.

georgweiss, 2020-03-24:
Well, my issue is not missing or incomplete menu bar. The issue is that the menu bar is unresponsive until I switch to another application and then back. I will give it a new try when #1230  is merged

kasemir, 2020-03-24:
It's responsive for me. In the past, I have seen an empty menu bar that only populated after switching between applications. Don't remember a populated but unresponsive one. Mind you that's still with OS 10.13.

georgweiss, 2020-03-24:
As stated earlier, I'm seeing this on Mac OS 10.15+ only. Worked fine with earlier versions.

georgweiss, 2020-11-13:
We're not alone: https://bugs.openjdk.java.net/browse/JDK-8233678

kasemir, 2020-11-13:
Good find. That gives us two options.

We can update the application code to remove the `setUseSystemMenuBar(true)` section:
https://github.com/ControlSystemStudio/phoebus/blob/b6325400c3e92fbe69b7f780f8ca90d705a00b5b/core/ui/src/main/java/org/phoebus/ui/application/PhoebusApplication.java#L391-L393

As a result, the menu bar becomes part of the window, as it has always been on MS Windows and Linux. Might be unexpected to Mac users, though.

The other option is to wait for https://bugs.openjdk.java.net/browse/JDK-8233678 to get fixed.

georgweiss, 2020-11-13:
Well, the bug has not been fixed in over a year... I'd say we set a deadline (say March 1st) and if not fixed by then we go for the proposed change.

georgweiss, 2020-12-10:
Another interesting observation: when launch from dmg built through jpackage, the application menu works as expected, i.e. user need not switch to another application and back. The window title bar also looks different compared to launching in different manner, see attached:
[image]
[image]

kasemir, 2020-12-10:
That is curious.. Suggests that there's no issue with our code, but we're missing some key parameters when starting the JVM.
Does Help, About, Details.., System Properties show any difference?
Also possible that the binary launcher created by jpackager, 
https://github.com/ControlSystemStudio/phoebus/wiki/Packaging,
does some things which can't be reproduced in a shell start script.

georgweiss, 2020-12-10:
I'm currently exploring jpackage and hope to find some answers.

georgweiss, 2020-12-10:
And by the way, there is no need to create the dmg to get the application menu to work. In the first step the .app folder is created, so opening this to launch will produce the same result.

georgweiss, 2020-12-10:
Sorry, the difference in window title bar is due to Java versions. OpenJDK 15 will render the dark version, but application menu issue is still there.

georgweiss, 2021-03-01:
Seems JavaFX 16 fixes the menu issue. Confirmed on JavaFX EA and JRE 11.

kasemir, 2021-03-01:
.. and Java 16 is to be released real soon. Will FX16 follow the same schedule?

georgweiss, 2021-03-01:
Do not know.

kasemir, 2021-03-01:
I don't see a definite date, either, but several promising hints.
The JavaFX 15 release in Sept 2020 was coupled to the Java SE 15 release.
For openjdk16, general availability is scheduled for March 16, ~2 weeks, and there is already a matching JavaFX 16 available as 'early access'.

So I suggest we wait for those ~2 weeks and then jump onto JavaFX 16.
If it's not available by the end of March, we use JavaFX 16 EA.

georgweiss, 2021-03-01:
My users have not complained about the menu issue, plus we're distributing installation packages that solves the issue anyway. So I'm in no hurry. Not sure I would vote for an EA release.

kasemir, 2021-04-22:
Fixed with JFX16

georgweiss, 2021-04-27:
Actually the problem still there when launching Phoebus from a unzipped package. However, it now works when running from IntelliJ, which it did not using previous JavaFX version.

kasemir, 2021-04-27:
With Mac OS 10.15.7, the SNS product from https://controlssoftware.sns.ornl.gov/css_phoebus/#SNS, which bundles JDK 16, starts fine with menu bar that's responding right away, no need to first focus another app.
Are you using Mac OS 11?

georgweiss, 2021-04-27:
Nope, 10.15.7. To clarify: as previously, installing Phoebus from a jpackage built installer is fine, no menu issues.

kasemir, 2021-04-27:
Interesting, same Mac OS and JDK. Does Help/About/.. show JavaFX Version "16+8"?

georgweiss, 2021-04-27:
Yes.

kasemir, 2021-04-27:
For me, the menu bar shows "Java" while the splash is up. Then the splash is hidden, the menu bar changes to Java, File, Applications, ... and reacts on the first click. Do you disable the splash?

georgweiss, 2021-04-27:
Same behavior here. I do not disable splash, but with -nosplash the menu still does not work.

georgweiss, 2021-04-27:
But this is not an issue here as we use jpackage.