ControlSystemStudio/phoebus #1094: Display Builder Runtime: log and e-mail context menu entries cannot be configured out

pavel-ch, 2020-01-31:
Since recent commits (not yet evaluating particular changes), the configuring out of Display Builder Runtime log and e-mail context menu entries does not work.
Entries are still shown, although should not due to the settings.ini statements:
```
# Intentionally empty
org.phoebus.logbook.ui/logbook_factory=
org.phoebus.email/mailhost=DISABLE
```
This worked for me before.

shroffk, 2020-01-31:
Hello, 
I feel the proper way to address this would be to 

1. Add actions to the context menu using SPI, selection service, adapter service and not hard code them in.
This would also mean that we can clean up the dependencies... Display builder would not depend on logbook and email. The actions for contributing to these would be part of a separate module.

2. Then in your product you choose to include what you need and what you don't. If you don't want logbook and email integration, leave out org.phoebus.displaybuilder.logbook & org.phoebus.displaybuilder.email. 
This is better than including a module, and then adding a preference to basically uninclude the module.

I have added this as a topic for the upcoming developers meeting and I will implement a solution based on the above description if you can wait for another 2 weeks.

georgweiss, 2020-01-31:
The fact of the matter is that we have been struggling with making the use of classes (fonts, colors...) behave in a predictable manner for some time before we realized that what we observe is a timing/race condition issue. We are determined to deploy a release next week where this works, so no, we cannot wait for another 2 weeks. We can cope with  @pavel-ch 's issue for the time being.
However, I like you approach and suggest you go ahead.

pavel-ch, 2020-01-31:
@shroffk  yes this looks fine. Waiting for a while is not an issue for me. Priority of this particular issue is low, however, having a more modular product is of a certain priority I think. Thanks.

kasemir, 2020-01-31:
> This worked for me before.

Sure? I see no change in the code.
In databrowser `Perspective.createContextMenu`, an `SendEmailAction` is always added.
Inside `SendEmailAction`, `EmailPreferences.isEmailSupported` is used to enable/disable.

Setting `org.phoebus.email/mailhost=DISABLE` thus disables the context menu entry, but doesn't remove it.

Same with `SendLogbookAction`.

kasemir, 2020-01-31:
The display builder checks `isEmailSupported` and `LogbookUiPreferences.is_supported` before adding the context menu items, so there they're indeed removed when not supported.

kasemir, 2020-01-31:
Kunal, I agree with your mid-term plan, but expect that to take a little longer. We have the basic `adapter` support to turn `<whatever>` into an email or log entry. But will also need some way to order the items, because the automagically added items are already causing problems.

See this context menu opened on a display widget.
It has a basic order:

1) Widget info dialog
2) PV-related entries: Open PV in probe, data browser, ...
3) Display related entries: Print, email or log screenshot, ...
...

The "Copy PV to Clipboard" entry should be on top.
Then in the middle of 2) there's "Create Log", no icon, doesn't do anything, and "Send Email", also no icon, opens empty email dialog. 
[image]

pavel-ch, 2020-01-31:
The context menu entries are indeed not included in:
org.csstudio.display.builder.runtime.app.ContextMenuSupport.fillMenu:203 - 206,
```
        if (EmailPreferences.isEmailSupported())
            items.add(new SendEmailAction(model_parent, "Display Screenshot", display_info, () ->  Screenshot.imageFromNode(model_parent)));
        if (LogbookUiPreferences.is_supported)
            items.add(new SendLogbookAction(model_parent, "Display Screenshot", display_info, () ->  Screenshot.imageFromNode(model_parent)));
```
but they are included with:
```
org.csstudio.display.builder.runtime.app.ContextMenuSupport.fillMenu:155:
            ContextMenuHelper.addSupportedEntries(node, menu);
```
which comes from `org.phoebus.ui.application.ContextMenuService.listSupportedContextMenuEntries()`

Deleting app-email from my product helps, but deleting app-logbook* does not because of dependency of  the `org.csstudio.display.builder.runtime.app.ContextMenuSupport` class on `org.phoebus.logbook.ui.LogbookUiPreferences` -> exception, no context menu at all.

kasemir, 2020-03-13:
I don't think there's been any progress on the adapter handling.
Should those `AdapterFactory` entries that result in erroneous "Create Log" and "Send Email" entries be disabled until the underlying framework is good enough? We'd still have the direct additions based on `is*Supported` which properly show/hide, have an icon, and provide the basic functionality.

shroffk, 2020-03-13:
Ok,
Removed them for the time being.

closed 2020-03-16 by pavel-ch