ControlSystemStudio/phoebus #1116: Drag-n-dropped files removed from the file browser

georgweiss, 2020-02-06:
When dragging and dropping an image (or other?) file from the file browser into the display editor, the file is removed from the file browser view. Closing and re-opening the parent folder does nothing. However, navigating away using the directory path field, and then back again, makes the file reappear.

kasemir, 2020-02-06:
That's because in #855 you changed the default mode from copy to move. The file browser is thus told that the file was (re)-moved when it got dropped into the display editor.
You can press the modifier key before starting the drag to get a copy.

georgweiss, 2020-02-07:
OK. I will take a look if this can be fixed without changing the expected behavior when moving files around in the file browser.

closed 2020-02-07 by georgweiss