ControlSystemStudio/phoebus #1117: local pv's of type boolean

shroffk, 2020-02-06:
A local pv that is of type boolean is quite useful as a flag or the place holder for a conditional check.

It would be nice to have support for 
`loc://my_flag<VBoolean>(false)`

kasemir, 2020-02-06:
Slight issue is that this is not a type that we get from IOCs.
Because of Channel Access, we get a VEnum for 'boolean' records.
PV Access is the same.

So most tools like the display don't handle VBoolean.
A Checkbox widget for example doesn't handle VBoolean.

For the UI, that actually makes some sense because we don't want "true/false" in the display but "open/close", "close/open", "off/on", "good/bad", "bad/OK", ....

Still, I agree that going forward adding support for VBoolean is useful.
With the latest changes we handle bool for local PVs, PV Access.

Still need to add support for it in the display widgets (as applicable) and the ScriptUtil.

shroffk, 2020-02-06:
The probe writes 0 and 1 as strings... so the write method for the local pv receives a string "0"/"1"
Both these string are then parsed by the boolean parser as false.

Ideally, it would be nice to be able to write both "true"/"false" & 0/1

In the ValueHelper, when the instance is of type String, I can try to parse it as a Number first and then try to parse it as true/false... but I am not sure this is the nicest solution.

kasemir, 2020-02-06:
I saw that when playing with the new `loc://xx<VBoolean>` earlier today, but wasn't sure what to do. Same problem actually with enums.

In principle you can define an enum with labels `[ "5", "4", "3", "2", "1" ]`.
When you now enter "1" into probe, did you mean to write the label "1" (ordinal 4), or ordinal 1 (label "4")?!

You can also have an enum with labels `[ "true", "false" ]` and now the ordinal 1 is "false".
So I decided to punt the issue on VBoolean and require entering "true" "false".
But I agree that these are pathological examples, so if you prefer to have the local PV accept "0", "1" for VBoolean, that's fine.

kasemir, 2020-02-06:
Technically, in the loc PV `ValueHelper.adapt` method, I would first check for "true", "false", and then fall back to trying to get a number, on success using `number != 0`

```
if (type == VBoolean)
{
  ...
  if (new_value instanceof String)
  {
     if ("true".equalsIgnoreCase(new_value))
         .. true ..
     else if ("false".equalsIgnoreCase(new_value))
        .. false ..
     else
        try
        {
            boolean value = Integer.parseInt(new_value) != 0;
        } 
        ...
```

closed 2021-03-10 by shroffk