ControlSystemStudio/phoebus #1124: Channel Table: show number of results

mark0n, 2020-02-06:
Would it be possible to show the number of PVs in the search result? We are frequently facing questions like

> If we start archiving all PVs matching a certain pattern how many PVs are we adding to the archiver? Do we need additional resources on the archiver?

or

> How many turbo pumps do we have in segment X?

These questions could be answered more efficiently if the number of elements in the result would be displayed in the status bar.

shroffk, 2020-02-07:
Hmm.. I can look into this.
There is a query limit (10,000) to ensure that searching for * does not overwhelm the client. I can report the number of results if they are less than the limit... otherwise say result >10000
 
There is support for scrolling in the new channelfinder which allows for a nice way to run large queries.

mark0n, 2020-02-07:
Good point. With this limit it's even more important to see the number of items. The result might actually be incomplete.

shroffk, 2021-01-27:
completed

closed 2021-01-27 by shroffk