ControlSystemStudio/phoebus #1126: Alarm Log Table: disable automatic updates

mark0n, 2020-02-06:
The Alarm Log Table looks like a great tool for troubleshooting issues. I had a hard time using it though: I had to sort (which works flawlessly). Due to the amount of detail in this table I need a couple of minutes to make sense of the data. Unfortunately, the table automatically updates every few seconds which can be quite distracting and might move the results I care about around. To make matters worse whenever the table updates it loses its search properties forcing me to resort the table.

Have you considered a feature allowing users to disable automatic updates (like a checkbox next to the search button)?

tanviash, 2020-02-07:
@shroffk Hi Kunal, I can work on getting this checkbox to disable auto updates of the table. Do you have any other suggestions that can work better than check-box?

shroffk, 2020-02-07:
> To make matters worse whenever the table updates it loses its search properties forcing me to resort the table.

I think we might want to address this... we should save the search and the sorting features of the alarm table.

> disable automatic updates

We can add a checkbox, however I think the above feature might be more important :)

closed 2020-02-07 by shroffk