ControlSystemStudio/phoebus #1134: Formula Functions

shroffk, 2020-02-11:
Add the formula functions from the eclipse cs-studio to the new core-pv.

[X] Extend ‘FormulaFunction’ to describe arguments, so that auto-completion can be supported. Idea: replace `getArgumentCount()` returning 2 with `getArguments()` returning `[ “base”, “exponent” ]`
[KK] Implement auto-completion of function names, showing their arguments
[ ] re implement as many of the formula functions as possible. Contribute using the new formula function SPI in core
[ ] Create a utility view for all the formula functions. This is useful for users.
[ ] Add documentation

kasemir, 2020-02-11:
`getArguments()` returning `[ "arg1", "..." ]` means: Needs one argument, plus maybe more.
Just `[ "arg1", "..." ]` means: Zero or more args.

kasemir, 2020-02-11:
Please create PR for added category soon, since that impacts everybody who's using the SPI

shroffk, 2020-02-11:
I thought we should work on this branch and then make a PR

shroffk, 2020-02-11:
I will create a PR and then keep this branch open for us to co-ordinate our work on

shroffk, 2020-02-13:
Hello,
It seems like with the use of String... in concat and highestSeverity, we run into some issues with Scanner. 
The scanner does not like "..." it breaks out of the argument parsing and then throws an exception since it is finding a "." instead of the ")"

@kasemir could you advise on the best way to clean this up

georgweiss, 2020-02-13:
The parsing of a formula like rescale(`sim://const(1,2,3)`, 20, 0) will call the formula implementation multiple times. The first time the first argument is set to a default VDouble NaN value. Second time it is called with the correct type and value.  The consequence is that checking for argument types becomes tricky. If the first argument is not an array, the implementation can throw an exception, but that would be logged. Or the implementation can check type and return null if it is not the expected, but ultimately this results in the value not being computed without any further information (except possibly in the log).

Any thoughts on this?

georgweiss, 2020-02-13:
Another thing: formulas are registered by name only, which does not allow for "method overloading". We could maybe append argument list to the key?

kasemir, 2020-02-13:
Kunal: Do you mean scanner for parsing formula for evaluation, or parser for completion proposals? When I asked about `concat` in yesterday's hangout, you sounded as if it basically evaluates, so I assume the issue in in the proposal lookup? That was implemented when there was no actual var-arg function. Should have implemented the concat while working on proposals...

kasemir, 2020-02-13:
Georg: Formulas can be invoked with changing input types. Channels can connect/disconnect, so fundamentally implementation needs to handle changing data types.

georgweiss, 2020-02-13:
OK, that's fine, but what should the formula implementation return if the arguments are of wrong type?
And: what about overloading functions using the argument list to create the key (and not only name)?

kasemir, 2020-02-13:
> what should the formula implementation return if the arguments are of wrong type?

A function that usually returns a number can return Double.NAN with invalid alarm.
A function that usually returns a String can return an empty string with invalid alarm, or an error as a string with invalid alarm.

georgweiss, 2020-02-13:
A function returning an array should then maybe return a one-element array with Double.NaN?

kasemir, 2020-02-13:
Kunal:  The FormulaFunction javadoc 
```
    /** Describe arguments
     *
     *  Special argument "..." is used to indicate that a variable
     *  number of arguments may follow.
     *  If it is used, it must be the last one in the list.
     *
     *  @return Description of arguments, for example `[ "base", "exponent" ]` */
    public List<String> getArguments();
```

was meant to suggest that "..." is itself shown as an argument, i.e. `concat` (why is in in `class StringFunction` as opposed to e.g. `class Concat`, do you expect to only have one string function?) should not return
```
 List.of("String...");
```
but 
```
 List.of("String", "...");
```

But maybe `String...` is better because it suggests: Zero or more args, using strings" while `String, ...` suggests: At least one string, and then maybe more arguments.
So I'll change the javadoc and the parsers.

kasemir, 2020-02-13:
Kunal: Missed that you added `isVarArgs()`, so the "..." string can be placed anywhere, it doesn't have any significance. I take that to mean: The `getArguments().size()` is the minimum number of arguments, and `isVarArgs()==true` means there could be more. The "..." can be placed anywhere, it has no function other than for the user to look at.

kasemir, 2020-02-13:
Actually, the way you call `isVarArgs()` you completely ignore the arg list, making no distinction between one-arg-or-more or zero-args-or-more. Fine, I'll go with that.

georgweiss, 2020-02-15:
The strategy on how to set the alarm. time and display fields of the formula function return value differs a bit among implementations. I would prefer if we can agree on some strategy, e.g. always Alarm.none(), Time.now(), Display.none().

georgweiss, 2020-02-17:
Would appreciate if someone could verify the ArrayRangeOf function as the old (diirt) implementation uses classes that are not available in the vtype lib.

georgweiss, 2020-02-17:
There is in the diirt implementation of formula functions the "DynamicFormulaFunction", which connects to PVs. Should this be supported in Phoebus?

shroffk, 2020-03-31:
> There is in the diirt implementation of formula functions the "DynamicFormulaFunction", which connects to PVs. Should this be supported in Phoebus?

No

I think we have most of the important ones... we should close this ticket and open it if we find anything missing.

georgweiss, 2020-08-13:
Long time, no see...
Apparently my implementation of array functions have not been subject to any PR, cannot say why. Any reason we should not integrate those?

shroffk, 2020-08-13:
Is there an open PR?

I started re implementing some of the array formula in relation to an auto scale issue.
Can you open a PR please

georgweiss, 2020-08-13:
I can create a PR early next week, is that OK?

georgweiss, 2020-08-13:
PR created

closed 2020-08-13 by shroffk