ControlSystemStudio/phoebus #1136: Display model PathTest.java fails in windows

DominicOram, 2020-02-11:
As of https://github.com/shroffk/phoebus/issues/991 the tests in PathTest.java fail on windows with:
```
[ERROR] Tests run: 4, Failures: 2, Errors: 0, Skipped: 0, Time elapsed: 0.002 s <<< FAILURE! - in org.csstudio.display.builder.PathTest
[ERROR] testRelative(org.csstudio.display.builder.PathTest)  Time elapsed: 0.001 s  <<< FAILURE!
java.lang.AssertionError:

Expected: "other.bob"
     but: was "..\other.bob"
        at org.csstudio.display.builder.PathTest.testRelative(PathTest.java:77)

[ERROR] testCombine(org.csstudio.display.builder.PathTest)  Time elapsed: 0 s  <<< FAILURE!
java.lang.AssertionError:

Expected: "/home/beamline/main.bob"
     but: was "https://webopi.sns.gov/webopi/opi//home/beamline/main.bob"
        at org.csstudio.display.builder.PathTest.testCombine(PathTest.java:67)

[INFO]
[INFO] Results:
[INFO]
```

DominicOram, 2020-02-11:
The linked PR allows the tests to run OK on both windows and linux without reducing code coverage but I'm not sure if it actually requires more thought as to how the functions behave?

kasemir, 2020-02-11:
It's definitely progress to have the test pass on Windows, even if that requires two different tests.

The requirements of the ModelResourceUtil are in no small part "Needs to work with deployed displays", where buttons, related displays etc. need to work relative to local files but also when used with "http://server/displays/some/display.opi". The same displays can be used both as files at a beam line, and via URLs when viewed from outside the beamline. So can't use file tools because the paths may refer to URLs, at the same time cannot use URL tools because the paths may not be valid URLs. For example, a space " " is OK in a file path. For  URL it needs to turn into "+" or some other URL escape, which then won't work when used as a file name.

Adding tests and having them pass is good, since in case we eventually decide to re-implement the ModelResourceUtil, those tests are the best description of how it needs to function. So thanks for fixing this on Windows.

DominicOram, 2020-02-12:
Ok, neither the " " or "+" cases are being tested in the unit tests so I will add them. Are you saying it's possible to have a mix of URL and path in the same string? Otherwise you could pass the tests by calling `isUrl` then running the appropriate URL relativize or Path relativize.

DominicOram, 2020-02-12:
Linked PR adds unit tests to cover these cases and continues to pass all tests using `nio` functions. Are there additional unit tests that you are worried the new implementation will miss? I can likely make similar gains in other functions like `normalize`

kasemir, 2020-02-12:
> Are you saying it's possible to have a mix of URL and path in the same string? 

Not sure how to explain, but here's an example:

You have display `/home/controls/beamline1/main.opi`. It contains a link (embedded display or button or nav tab or ...) to `../shared displays/motor/single motor detail.opi`.

That screen is opened two ways:

1) At the beamline, where the actual files are, by opening `/home/controls/beamline1/main.opi`. The link is resolved into `/home/controls/shared displays/motor/single motor detail.opi`.

2) From the office, using a CA gateway for read-only access to the data and fetching the display files from a web server by opening `http://display.files.site.org/beamline1/main.opi`, where the link now needs to resolve as `http://display.files.site.org/shared displays/motor/single motor detail.opi`.

In case 1, there's the added quirk on Windows that everything "works" with `/` as a path separator, meaning you can open files just fine, but when you use filename tools like `File` or `Path` they tend to change the path separators into backslashes.

Similar in case 2, you can for the most part use `http:// .. with spaces ..` but to have a valid URL/URI those spaces need to be escaped.

Finally, we like to be able to compare paths. When you open the same display "again", the one that's already open should pop up, we don't want to open N instances of the same display. We try this by comparing the 'input' of a `DockItemWithInput` against a path that we're about to open. having a 'normalized' version of the path helps with this.

kasemir, 2020-02-12:
One aspect of all of this: In the display editor, we want to create relative links.

Assume you edit "main.opi" or "main.bob" and then configure the path to a link (embedded display, button, nav tab, PNG to show in picture widget, ...). You push that "..." button to get a file browser, navigate to `/home/controls/shared displays/motor/single motor detail.opi`. 
Thar absolute path only works right now where you edit the file, while a relative path will also work via `http://..` access, or when you move the root of the complete file tree from `/home/controls` to another location.
So we don't want to save that absolute path, instead turn it into a relative path `../shared displays/motor/single motor detail.opi`.

We thus need something like `getRelativePath` or `relativizePaths`. It should return something that will work on Linux, Mac, Windows. You might be editing that file in your office on a Windows computer, but we cannot save the file with a backslash-path  `..\shared displays\motor\single motor detail.opi` since a file saved with backslashes won't work when the display is later used at the beamline under Linux.

DominicOram, 2020-02-12:
Great, thanks for this, it makes it a lot clearer. I will add more documentation and tests to match these use cases then see if I can simplify with nio whilst still passing them.

kasemir, 2020-03-17:
Can we close this?

kasemir, 2020-03-18:
Commit messages suggest this has been fixed. Re-open in case issue persists.

closed 2020-03-18 by kasemir