ControlSystemStudio/phoebus #1155: Save and Restore service won't start up

geniejhang, 2020-02-17:
Hi,

I'm trying to start-up the save and restore service with the latest commit ad534272ad1204eced6e25d732513a47c57765c0
and it gives an error `Migration V19_0__migrate_pv_data_step3.sql failed`.

I leave the full error message below.

```
  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::        (v2.1.5.RELEASE)

2020-02-17 15:40:21.348 [main] INFO  org.phoebus.service.saveandrestore.application.Application - Starting Application on lamp-debian with PID 14626 (/home/changj/IdeaProjects/phoebus_github/services/save-and-restore/target/service-save-and-restore-4.6.3-SNAPSHOT.jar started by changj in /home/changj/IdeaProjects/phoebus_github/services/save-and-restore/target)
2020-02-17 15:40:21.354 [main] INFO  org.phoebus.service.saveandrestore.application.Application - No active profile set, falling back to default profiles: default
2020-02-17 15:40:22.506 [main] INFO  org.springframework.context.support.PostProcessorRegistrationDelegate$BeanPostProcessorChecker - Bean 'org.springframework.transaction.annotation.ProxyTransactionManagementConfiguration' of type [org.springframework.transaction.annotation.ProxyTransactionManagementConfiguration$$EnhancerBySpringCGLIB$$ad31f770] is not eligible for getting processed by all BeanPostProcessors (for example: not eligible for auto-proxying)
2020-02-17 15:40:22.763 [main] INFO  org.springframework.boot.web.embedded.tomcat.TomcatWebServer - Tomcat initialized with port(s): 8080 (http)
2020-02-17 15:40:22.802 [main] INFO  org.apache.coyote.http11.Http11NioProtocol - Initializing ProtocolHandler ["http-nio-8080"]
2020-02-17 15:40:22.814 [main] INFO  org.apache.catalina.core.StandardService - Starting service [Tomcat]
2020-02-17 15:40:22.815 [main] INFO  org.apache.catalina.core.StandardEngine - Starting Servlet engine: [Apache Tomcat/9.0.19]
2020-02-17 15:40:22.917 [main] INFO  org.apache.catalina.core.ContainerBase.[Tomcat].[localhost].[/] - Initializing Spring embedded WebApplicationContext
2020-02-17 15:40:22.917 [main] INFO  org.springframework.web.context.ContextLoader - Root WebApplicationContext: initialization completed in 1516 ms
2020-02-17 15:40:23.380 [main] INFO  org.flywaydb.core.internal.license.VersionPrinter - Flyway Community Edition 5.2.4 by Boxfuse
2020-02-17 15:40:23.391 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Starting...
2020-02-17 15:40:23.579 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Start completed.
2020-02-17 15:40:23.582 [main] INFO  org.flywaydb.core.internal.database.DatabaseFactory - Database: jdbc:mysql://localhost/saverestore_github (MySQL 5.5)
2020-02-17 15:40:23.709 [main] INFO  org.flywaydb.core.internal.command.DbValidate - Successfully validated 21 migrations (execution time 00:00.042s)
2020-02-17 15:40:23.749 [main] INFO  org.flywaydb.core.internal.schemahistory.JdbcTableSchemaHistory - Creating Schema History table: `saverestore_github`.`flyway_schema_history`
2020-02-17 15:40:23.792 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Name 'flyway_schema_history_pk' ignored for PRIMARY key. (SQL State: 42000 - Error Code: 1280)
2020-02-17 15:40:23.812 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Current version of schema `saverestore_github`: << Empty Schema >>
2020-02-17 15:40:23.812 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 1.0 - initial
2020-02-17 15:40:23.919 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 2.0 - add provider
2020-02-17 15:40:23.935 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 3.0 - add snapshot name
2020-02-17 15:40:23.955 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 4.0 - rename root node
2020-02-17 15:40:23.976 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 5.0 - alter snapshot pv simplify config pv
2020-02-17 15:40:24.104 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 6.0 - remove unused columns
2020-02-17 15:40:24.158 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 7.0 - add usernameid to node
2020-02-17 15:40:24.186 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 8.0 - snapshot as node
2020-02-17 15:40:24.206 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Trigger does not exist (SQL State: HY000 - Error Code: 1360)
2020-02-17 15:40:24.210 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Trigger does not exist (SQL State: HY000 - Error Code: 1360)
2020-02-17 15:40:24.292 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 9.0 - add golden to snapshot
2020-02-17 15:40:24.302 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 10.0 - add unique id to node
2020-02-17 15:40:24.327 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 11.0 - set unique id on node
2020-02-17 15:40:24.357 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 12.0 - add properties to node
2020-02-17 15:40:24.377 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 13.0 - set root property
2020-02-17 15:40:24.394 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 14.0 - add readback
2020-02-17 15:40:24.445 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 15.0 - remove provider
2020-02-17 15:40:24.463 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 16.0 - update config pv unique constraint
2020-02-17 15:40:24.478 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 17.0 - migrate pv data step1
2020-02-17 15:40:24.501 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 18.0 - migrate pv data step2
2020-02-17 15:40:24.509 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `saverestore_github` to version 19.0 - migrate pv data step3
2020-02-17 15:40:24.510 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Key column 'name' doesn't exist in table (SQL State: 42000 - Error Code: 1072)
2020-02-17 15:40:24.511 [main] ERROR org.flywaydb.core.internal.command.DbMigrate - Migration of schema `saverestore_github` to version 19.0 - migrate pv data step3 failed! Please restore backups and roll back database and code!
2020-02-17 15:40:24.516 [main] WARN  org.springframework.boot.web.servlet.context.AnnotationConfigServletWebServerApplicationContext - Exception encountered during context initialization - cancelling refresh attempt: org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'configurationController': Unsatisfied dependency expressed through field 'services'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'services': Unsatisfied dependency expressed through field 'nodeDAO'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'configDAO': Unsatisfied dependency expressed through field 'jdbcTemplate'; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'flywayInitializer' defined in class path resource [org/springframework/boot/autoconfigure/flyway/FlywayAutoConfiguration$FlywayConfiguration.class]: Invocation of init method failed; nested exception is org.flywaydb.core.internal.command.DbMigrate$FlywayMigrateException: 
Migration V19_0__migrate_pv_data_step3.sql failed
[... cut ...]
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.doResolveDependency(DefaultListableBeanFactory.java:1248) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.resolveDependency(DefaultListableBeanFactory.java:1168) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.inject(AutowiredAnnotationBeanPostProcessor.java:593) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        ... 27 more

```

geniejhang, 2020-02-17:
@tanviash @georgweiss 
Sorry. Forgot to tag people.

georgweiss, 2020-02-17:
I have a few days off, will check this on Friday Feb 21st.

georgweiss, 2020-02-21:
I cannot reproduce this (using Mysql 8.0 on Mac OS), tried bot existing database and new one. Can you pleas check the structure of the config_pv table? Stack trace suggests the column "name" does not exist,  but it should.

geniejhang, 2020-02-21:
I tried just now to build everything with a newly cloned, latest repository and empty database.
Is it because I'm using MariaDB? 

I confirmed there's the name column in config_pv table.
Also, tried to execute the query in V19.0~.sql directly.

```
changj@lamp-debian:~/IdeaProjects/phoebus_github/services/save-and-restore$ mysql -u phoebus -p
Enter password: 
Welcome to the MariaDB monitor.  Commands end with ; or \g.
Your MariaDB connection id is 876
Server version: 10.3.22-MariaDB-0+deb10u1 Debian 10

Copyright (c) 2000, 2018, Oracle, MariaDB Corporation Ab and others.

Type 'help;' or '\h' for help. Type '\c' to clear the current input statement.

MariaDB [(none)]> use saverestore_github;
Reading table information for completion of table and column names
You can turn off this feature to get a quicker startup with -A

Database changed
MariaDB [saverestore_github]> show tables;
+------------------------------+
| Tables_in_saverestore_github |
+------------------------------+
| config_pv                    |
| config_pv_relation           |
| flyway_schema_history        |
| global_sequence              |
| node                         |
| node_closure                 |
| property                     |
| pv                           |
| snapshot_node                |
| snapshot_node_pv             |
+------------------------------+
10 rows in set (0.000 sec)

MariaDB [saverestore_github]> describe config_pv;
+----------------+---------------------+------+-----+---------+----------------+
| Field          | Type                | Null | Key | Default | Extra          |
+----------------+---------------------+------+-----+---------+----------------+
| id             | bigint(20) unsigned | NO   | PRI | NULL    | auto_increment |
| name           | varchar(50)         | NO   | UNI | NULL    |                |
| readback_name  | varchar(50)         | YES  |     | NULL    |                |
| readonly       | tinyint(1)          | YES  |     | 0       |                |
| pv_id          | int(11)             | YES  |     | NULL    |                |
| readback_pv_id | int(11)             | YES  |     | NULL    |                |
+----------------+---------------------+------+-----+---------+----------------+
6 rows in set (0.001 sec)

MariaDB [saverestore_github]> ALTER TABLE config_pv DROP COLUMN name;
ERROR 1072 (42000): Key column 'name' doesn't exist in table
MariaDB [saverestore_github]> 
```

georgweiss, 2020-02-21:
Well, I have not worked with MariaDB at all, so I have not performed any testing. For the moment I need to ask for your patience if you want me to verify on MariaDB. Meanwhile, can you test using Mysql or Postgresql?

geniejhang, 2020-02-21:
I confirmed everything works fine with mySQL 8.0 on Debian buster.
I thought MariaDB and mySQL behave the same, but probably not anymore.

georgweiss, 2020-02-21:
I would have made the same assumption. Can you continue using Mysql? I'm asking since I would need to postpone MariaDB verification for some time. And I would actually prefer to use Hibernate internally to get reduce work on DB engine verification.

georgweiss, 2020-02-24:
Can we close this?

geniejhang, 2020-02-24:
Yes, Sure! Thank you for your comments!

geniejhang, 2020-03-30:
@georgweiss @tanviash  I found why flyway migration fails with simple column dropping query:
`ALTER TABLE config_pv DROP COLUMN name;`

It's explained in [MariaDB Drop Column](https://mariadb.com/kb/en/alter-table/) as below.
> Dropping a column that is part of a multi-column UNIQUE constraint is not permitted.

So, adding the following line before `DROP COLUMN` query solves the issue with MariaDB.
`ALTER TABLE config_pv DROP INDEX config_pv_name_key;` 

Except for the migration, add/remove/modify record should be the same as mysql resulting in the same performance with mysql jdbc driver.

The real problem is that if V19 migration file is modified, migration checksum stored in DB doesn't match crashing the service.

If you don't want the line in, we can fork our own version for supporting MariaDB.
Thanks.

georgweiss, 2020-03-30:
I would maybe check with the Flyway documentation if there is a different way to solve this, e.g. recover from failed migration or trap a failed script. Unfortunately I cannot investigate this for the coming 4-5 weeks or so.

georgweiss, 2020-03-31:
OK, I realize I need to do this since geniejhang does not seem to have write access to this project. What I need is the jdbc driver class name for MariaDB.

geniejhang, 2020-03-31:
I tried with Maven module for [MariaDB jdbc](https://mvnrepository.com/artifact/org.mariadb.jdbc/mariadb-java-client/2.6.0), which was not successful.
What we need is just add the following line
`ALTER TABLE config_pv DROP INDEX config_pv_name_key;`

to the [V19_0__migrate_pv_data_step3.sql](https://github.com/ControlSystemStudio/phoebus/blob/master/services/save-and-restore/src/main/resources/db/migration/mysql/V19_0__migrate_pv_data_step3.sql) above 
`ALTER TABLE config_pv DROP COLUMN name;`.

I don't think masar service's sql queries are such complex ones that makes changes in the results between mariadb and mysql.
Using mysql jdbc is just fine once tables are correctly migrated.
I checked the table structure in both mariadb and mysql after adding the line. They were identical.

georgweiss, 2020-03-31:
OK, what I can do is to add migration script for MariaDB, but you will need to assist in the verification.

geniejhang, 2020-03-31:
Absolutely! Thanks.

georgweiss, 2020-03-31:
Please checkout branch CSSTUDIO-1233 and verify.

geniejhang, 2020-03-31:
This way definitely works because I already tested it.
However, to make sure, I tried by cloning the branch you specified and it works great.

Thanks.

georgweiss, 2020-03-31:
OK, I will create a pull request for the master branch.

closed 2020-03-31 by geniejhang