ControlSystemStudio/phoebus #1157: Setting MACROS when launching a display using display_runtime

douglasbeniz, 2020-02-20:
Hello guys,

I'm trying, without success, to set some MACROs on a display when launching it as described in the help:
`-resource file:/path/to/file?app=display_runtime&MACRO1=value+1&MACRO2=abc - Opens file with 'display_runtime' app, passing macros.`

The display is being opened, but no macro is being set.
--
I was able to do it, though, configuring a file called by -settings with this (e.g.):
`org.csstudio.display.builder.model/macros=<DEV>MB:PSU</DEV>`

However, using such approach is creating a file at:
`/<USER_HOME>/.phoebus/.userPrefs/org/csstudio/display/builder/model/prefs.xml`
and the MACRO is indefinitely set there even when previous settings file is not loaded. It is replaced when a new value is configured, yes, but I'd like to avoid that way, otherwise could be difficult to identify, if you don't know it was set before, from where such MACRO is being set.

--
I'm sorry if this is not the right channel to ask this, but I'd be grateful for any help!

Thanks in advance!

kasemir, 2020-02-20:
Can you try with just ` -resource file:/path/to/file?MACRO1=value+1`? When the file is opened, right-click and select the first ".. information" entry from the context menu, then check the "Macros" tab if "MACRO1" has been passed.

What might be happening is a bug in the launcher shell script that takes the first `&` as a "send-to-background", instead of passing the complete `file:/path/to/file?app=display_runtime&MACRO1=value+1&MACRO2=abc`

douglasbeniz, 2020-02-21:
Hey, @kasemir , thanks for the quick answer, firstly! Yes, you're correct, suppressing `app=display_runtime` it works properly! 

We will use it, so. Thank you!

douglasbeniz, 2020-02-21:
I'm sorry, @kasemir, I just noticed that if we need more MACROS, putting that `&` to concatenate them is failing as you said, probably because is being identified as a command to sent to background...

So, it works if we pass only 1 MACRO.

kasemir, 2020-02-21:
Linux 101: Enclose the parameters in quotes to keep the shell from handling the `&`:
```
phoebus.sh  -resource "file://path/to/01_main.bob&MACRO1=value+1&MACRO2=abc"
```

douglasbeniz, 2020-02-21:
That is it! Perfect! Using the double-quotes made it work even using the "app=display_runtime"!

I'm completely happy with that. Thank you!

closed 2020-02-21 by kasemir