ControlSystemStudio/phoebus #1161: Windows launcher broken since version update

kasemir, 2020-02-21:
#1102 updated the version to 4.6.1, but the Phoebus product windows launcher, https://github.com/ControlSystemStudio/phoebus/blob/master/phoebus-product/phoebus.bat, still uses 4.6.0.

georgweiss, 2020-02-24:
Changed to 4.6.1, but on the other hand that won't fix the problem if the build is for the next development version. Suggest this should be fixed using Maven filtering where the current version is written into the bat file during build. The subsequent IF statement could then be removed.

georgweiss, 2020-02-27:
Any comments @kasemir ? Close this?

kasemir, 2020-02-27:
Have you changed it to using Maven filtering as you suggested?

georgweiss, 2020-02-27:
No, if you want it I will do it.

kasemir, 2020-02-27:
What we really need is a windows launcher beyond just a batch file.
On Linux, shell scripts are fine, even expected.
For the Mac, we have a proper `*.app` wrapper.
Windows users are often the most clueless and have a hard time finding the `*.bat` file, especially when you can't even see the `*.bat` file ending.

Still, while all we have is the `*.bat` file, automatically setting the version number in there would be one improvement. But that could break when you auto-update to the next version.
So better than having maven place one version number in there might be a mechanism as in the shell scripts which use something like `JAR=`echo ${TOP}/product-*.jar`` to locate the jar file, matching whatever version there is.

georgweiss, 2020-02-27:
I will not fix the auto-update issue.
What we're discussing here is to use Microsoft tools to build MSI installers. The idea is not to support auto-update, but the MSI mechanism may prompt the user if the existing installation should be overwritten (possibly excluding a settings file). This way users may install/update by just clicking on a MSI file on the network.
Right now I am assuming the community version of Visual Studio has the tools we need to achieve this.

closed 2020-03-02 by kasemir