ControlSystemStudio/phoebus #1168: Unsupported widget 'org.csstudio.channel.opiwidgets.ChannelViewer'

tanviash, 2020-02-25:
@shroffk Hi, I found this one unsupported widget type in DisplayBuilder.

2020-02-25 11:30:52 SEVERE [org.csstudio.display.builder.model] Unknown widget type 'org.csstudio.channel.opiwidgets.ChannelViewer', line:91              note: each unknown widget type is reported only once for each model it appears in
2020-02-25 11:30:52 SEVERE [org.csstudio.display.builder.model] There were 1 error(s) during loading display from stream

kasemir, 2020-02-26:
Related:
At the workgroup meeting there was a question how to document the porting process, like a place where we can add details for specific widgets etc.

Started that in the wiki,  https://github.com/ControlSystemStudio/phoebus/wiki/Display-Builder-Compatibility

shroffk, 2020-02-26:
Hi @tanviash 
I don't have any plans on moving the ChannelViewer widget.

Do you have many screens that actually use this widget?

shroffk, 2020-02-28:
@tanviash any luck confirming if these widgets are essential.
I think in most cases using the channel table or channel tree as a seperate app is better.

tanviash, 2020-03-09:
Hi, we will be modifying the opi file in order to replace or remove the unsupported channelviewer widget since it was just in just 1 opi file. Closing this issue.

closed 2020-03-09 by tanviash