ControlSystemStudio/phoebus #1169: Unable to find "Show Channel Info" and "Modify Channel" in PV context menu

geniejhang, 2020-02-25:
I'm pressing buttons here and there and comparing stuff.
Today, I was trying to look into the details of a channel and found something looks missing.

The upper figure is a screenshot of CS-Studio 4.6 and the bottom one is of Phoebus.
As you can see, Two items, "Show Channel Info" and "Modify Channel" are missing.

Is that because I'm not including something for them when building the package?
I see that I have `app-channel-views`, `app-channel-channelfinder`, and `app-channel-utility` in `pom.xml` file in `phoebus-product` folder.

Thanks.

[image]
[image]

shroffk, 2020-02-26:
I think the modify channel action is not very useful. You will rarely update many tags and properties on a channel or channels simultaneously... and even in the rare cases where you would like to do this a python script might be a better than a UI dialog.

I will add a command akin to "show channel info"

geniejhang, 2020-02-27:
Okay. Thanks.
I just wonder if it's the designed behavior.

shroffk, 2020-02-28:
#1178
The above PR adds the show info feature, @geniejhang can you check and close this issue ticket :)

geniejhang, 2020-02-28:
That was pretty fast! I just confirmed it. Thanks! @shroffk

closed 2020-02-28 by geniejhang