ControlSystemStudio/phoebus #1170: Context menu not opened on first right click

georgweiss, 2020-02-26:
In some views the context menu is rendered only upon second right-click. For instance the display editor (right click in the layout editor), the PVTable, PVList and PVTree. There may be more examples, but the common factor for these examples is that the context menu is constructed in the onContextMenuRequested handler. In other cases (e.g. AlarmTableUI) the handler additionally calls menu.show(), which results in the context menu being rendered on first right click as expected.

I have not been successful in finding the cause of the observed behavior. I would maybe blame JavaFX, but I haven't found any posts related to this.

If the solution is to always call menu.show() I'll be happy to implement it.

kasemir, 2020-02-26:
Not sure if this is specific to opening context menus.
Just trying to close a tab by clicking on the 'x' in the tab often requires two clicks.

Maybe both the open-context-menu and the tab-close-issue are related to somehow dropping a mouse event?

georgweiss, 2020-02-26:
Well, the handler is called already on first click, but then the rest is up to JavaFX...

kasemir, 2020-02-26:
Do you always see the `onContextMenuRequested` call?
That's great, then the context menu issue is not about dropped mouse events, might indeed just need to force `menu.show()` right in there. Or maybe a `Platform.runLater(menu::show)`.
Go ahead and try that.

georgweiss, 2020-02-26:
Yes. This is 100% reproducible.
But if Platform.runLater is preferable we should use it also for those cases where menu.show() is already used.

georgweiss, 2020-02-26:
Not sure I see a need for Platform.runLater...

kasemir, 2020-02-26:
If you don't need `runLater`, don't use it. Was only suggesting in case this is a JavFX timing issue.

Now I wonder if the underlying issue is that the context menu is initially empty.
On the first onContextMenuRequested it's populated, but JavaFX doesn't show it because it was empty.
On the next onContextMenuRequested it's (maybe) updated, but since it was non-empty, JavaFX shows it.
There's a comment related to that in 
app/display/runtime ProbeDisplayContextMenuEntry.java:

``` *  When empty, context menu is not shown.```

kasemir, 2020-02-26:
When showing the context menu, there are different options.
I don't remember the detail, but I think this one worked best when running with multiple monitors:
```
setOnContextMenuRequested(event ->
            menu.show(this.getScene().getWindow(), event.getScreenX(), event.getScreenY())
```

georgweiss, 2020-02-26:
Could be the case, but then that would suggest that onRequestContextMenu cannot be used to configure the context menu. And wouldn't this mean that JavaFX calls the handler on a different thread than the thread rendering the menu? 

In (at least) the DisplayEditorInstance the handler does a lot of things, some of which are not dependent on the selection. One could maybe consider moving the non-dynamic object instantiation outside the handler?

A have already started on updating to include menu.show() as in your example.

closed 2020-02-26 by georgweiss