ControlSystemStudio/phoebus #1174: Kafka crash

georgweiss, 2020-02-27:
We have seen a Kafka crash as shown below. There is limited information out there related to this, so I would like query the community if anyone has seen something similar.

```
[2020-01-24 14:57:34,246] INFO [Log partition=ODHTalk-0, dir=/tmp/kafka-logs] Found deletable segments with base offsets [1] due to retention time 604800000ms breach (kafka.log.Log)
[2020-01-24 14:57:34,247] INFO [Log partition=ODHTalk-0, dir=/tmp/kafka-logs] Scheduling log segment [baseOffset 1, size 136532] for deletion. (kafka.log.Log)
[2020-01-24 14:57:34,250] ERROR Error while deleting segments for ODHTalk-0 in dir /tmp/kafka-logs (kafka.server.LogDirFailureChannel)
java.nio.file.NoSuchFileException: /tmp/kafka-logs/ODHTalk-0/00000000000000000001.log
	at java.base/sun.nio.fs.UnixException.translateToIOException(UnixException.java:92)
	at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111)
	at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:116)
	at java.base/sun.nio.fs.UnixCopyFile.move(UnixCopyFile.java:430)
	at java.base/sun.nio.fs.UnixFileSystemProvider.move(UnixFileSystemProvider.java:267)
	at java.base/java.nio.file.Files.move(Files.java:1421)
	at org.apache.kafka.common.utils.Utils.atomicMoveWithFallback(Utils.java:815)
	at org.apache.kafka.common.record.FileRecords.renameTo(FileRecords.java:224)
	at kafka.log.LogSegment.changeFileSuffixes(LogSegment.scala:508)
	at kafka.log.Log.asyncDeleteSegment(Log.scala:1962)
	at kafka.log.Log.deleteSegment(Log.scala:1947)
	at kafka.log.Log.$anonfun$deleteSegments$3(Log.scala:1493)
	at kafka.log.Log.$anonfun$deleteSegments$3$adapted(Log.scala:1493)
	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
	at kafka.log.Log.$anonfun$deleteSegments$2(Log.scala:1493)
	at scala.runtime.java8.JFunction0$mcI$sp.apply(JFunction0$mcI$sp.java:23)
	at kafka.log.Log.maybeHandleIOException(Log.scala:2065)
	at kafka.log.Log.deleteSegments(Log.scala:1484)
	at kafka.log.Log.deleteOldSegments(Log.scala:1479)
	at kafka.log.Log.deleteRetentionMsBreachedSegments(Log.scala:1557)
	at kafka.log.Log.deleteOldSegments(Log.scala:1547)
	at kafka.log.LogManager.$anonfun$cleanupLogs$3(LogManager.scala:914)
	at kafka.log.LogManager.$anonfun$cleanupLogs$3$adapted(LogManager.scala:911)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at kafka.log.LogManager.cleanupLogs(LogManager.scala:911)
	at kafka.log.LogManager.$anonfun$startup$2(LogManager.scala:395)
	at kafka.utils.KafkaScheduler.$anonfun$schedule$2(KafkaScheduler.scala:114)
	at kafka.utils.CoreUtils$$anon$1.run(CoreUtils.scala:65)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
	Suppressed: java.nio.file.NoSuchFileException: /tmp/kafka-logs/ODHTalk-0/00000000000000000001.log -> /tmp/kafka-logs/ODHTalk-0/00000000000000000001.log.deleted
		at java.base/sun.nio.fs.UnixException.translateToIOException(UnixException.java:92)
		at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111)
		at java.base/sun.nio.fs.UnixCopyFile.move(UnixCopyFile.java:417)
		at java.base/sun.nio.fs.UnixFileSystemProvider.move(UnixFileSystemProvider.java:267)
		at java.base/java.nio.file.Files.move(Files.java:1421)
		at org.apache.kafka.common.utils.Utils.atomicMoveWithFallback(Utils.java:812)
		... 29 more
[2020-01-24 14:57:34,253] ERROR Uncaught exception in scheduled task 'kafka-log-retention' (kafka.utils.KafkaScheduler)
org.apache.kafka.common.errors.KafkaStorageException: Error while deleting segments for ODHTalk-0 in dir /tmp/kafka-logs
Caused by: java.nio.file.NoSuchFileException: /tmp/kafka-logs/ODHTalk-0/00000000000000000001.log
	at java.base/sun.nio.fs.UnixException.translateToIOException(UnixException.java:92)
	at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111)
	at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:116)
	at java.base/sun.nio.fs.UnixCopyFile.move(UnixCopyFile.java:430)
	at java.base/sun.nio.fs.UnixFileSystemProvider.move(UnixFileSystemProvider.java:267)
	at java.base/java.nio.file.Files.move(Files.java:1421)
	at org.apache.kafka.common.utils.Utils.atomicMoveWithFallback(Utils.java:815)
	at org.apache.kafka.common.record.FileRecords.renameTo(FileRecords.java:224)
	at kafka.log.LogSegment.changeFileSuffixes(LogSegment.scala:508)
	at kafka.log.Log.asyncDeleteSegment(Log.scala:1962)
	at kafka.log.Log.deleteSegment(Log.scala:1947)
	at kafka.log.Log.$anonfun$deleteSegments$3(Log.scala:1493)
	at kafka.log.Log.$anonfun$deleteSegments$3$adapted(Log.scala:1493)
	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
[... 47 lines cut ...]
	at kafka.server.BrokerTopicMetrics.newMeter(KafkaRequestHandler.scala:143)
	at kafka.server.BrokerTopicMetrics.<init>(KafkaRequestHandler.scala:149)
	at kafka.server.BrokerTopicStats$.$anonfun$valueFactory$1(KafkaRequestHandler.scala:197)
	at kafka.utils.Pool.$anonfun$getAndMaybePut$1(Pool.scala:47)
	at kafka.utils.Pool$$anon$1.apply(Pool.scala:61)
	at java.base/java.util.concurrent.ConcurrentHashMap.computeIfAbsent(ConcurrentHashMap.java:1705)
	at kafka.utils.Pool.getAndMaybePut(Pool.scala:60)
	at kafka.utils.Pool.getAndMaybePut(Pool.scala:47)
	at kafka.server.BrokerTopicStats.topicStats(KafkaRequestHandler.scala:207)
	at kafka.server.ReplicaManager.read$1(ReplicaManager.scala:919)
	at kafka.server.ReplicaManager.$anonfun$readFromLocalLog$4(ReplicaManager.scala:1003)
	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
	at kafka.server.ReplicaManager.readFromLocalLog(ReplicaManager.scala:1002)
	at kafka.server.ReplicaManager.readFromLog$1(ReplicaManager.scala:852)
	at kafka.server.ReplicaManager.fetchMessages(ReplicaManager.scala:857)
	at kafka.server.KafkaApis.handleFetchRequest(KafkaApis.scala:757)
	at kafka.server.KafkaApis.handle(KafkaApis.scala:114)
	at kafka.server.KafkaRequestHandler.run(KafkaRequestHandler.scala:69)
	at java.base/java.lang.Thread.run(Thread.java:834)
[2020-01-24 14:57:34,343] INFO [ReplicaManager broker=0] Broker 0 stopped fetcher for partitions __consumer_offsets-22,TS2-PSSTalk-0,__consumer_offsets-30,__consumer_offsets-8,__consumer_offsets-21,LCRTalk-0,__consumer_offsets-4,TMCPTalk-0,TS2-PSSCommand-0,LCR-0,__consumer_offsets-27,__consumer_offsets-7,__consumer_offsets-9,__consumer_offsets-46,ODHTalk-0,TICP-0,ALARM-0,__consumer_offsets-25,__consumer_offsets-35,__consumer_offsets-41,__consumer_offsets-33,__consumer_offsets-23,__consumer_offsets-49,ACCP-0,__consumer_offsets-47,__consumer_offsets-16,__consumer_offsets-28,__consumer_offsets-31,__consumer_offsets-36,__consumer_offsets-42,__consumer_offsets-3,ACCPTalk-0,__consumer_offsets-18,__consumer_offsets-37,__consumer_offsets-15,__consumer_offsets-24,ODHCommand-0,TICPTalk-0,TS2Talk-0,__consumer_offsets-38,TS2-0,__consumer_offsets-17,LCRCommand-0,__consumer_offsets-48,__consumer_offsets-19,__consumer_offsets-11,ODH-0,ALARMCommand-0,TMCPCommand-0,__consumer_offsets-13,__consumer_offsets-2,__consumer_offsets-43,__consumer_offsets-6,__consumer_offsets-14,TS2Command-0,__consumer_offsets-20,__consumer_offsets-0,__consumer_offsets-44,TS2-PSS-0,ALARMTalk-0,__consumer_offsets-39,__consumer_offsets-12,__consumer_offsets-45,__consumer_offsets-1,__consumer_offsets-5,__consumer_offsets-26,TICPCommand-0,__consumer_offsets-29,__consumer_offsets-34,ACCPCommand-0,__consumer_offsets-10,TMCP-0,__consumer_offsets-32,__consumer_offsets-40 and stopped moving logs for partitions  because they are in the failed log directory /tmp/kafka-logs. (kafka.server.ReplicaManager)
[2020-01-24 14:57:34,343] INFO Stopping serving logs in dir /tmp/kafka-logs (kafka.log.LogManager)
[2020-01-24 14:57:34,346] ERROR Shutdown broker because all log dirs in /tmp/kafka-logs have failed (kafka.log.LogManager)
```

kasemir, 2020-02-27:
> java.nio.file.NoSuchFileException: /tmp/kafka-logs/ODHTalk-0/00000000000000000001.log

You can't run a production setup with files placed in /tmp.
The OS will delete them.

See https://github.com/ControlSystemStudio/phoebus/blob/master/app/alarm/Readme.md:

```
Check config/zookeeper.properties and config/server.properties. By default these contain settings for keeping data in /tmp/, which works for initial tests, but risks that Linux will delete the data. For a production setup, change zookeeper.properties: ...
```

georgweiss, 2020-03-02:
Thanks, will update the log path

closed 2020-03-02 by georgweiss