ControlSystemStudio/phoebus #1177: Expected behavior in Save & restore app?

geniejhang, 2020-02-27:
@georgweiss Once a PV is added to DB with "Read Only" flag,
when that PV is added to a different saveset, user cannot change "Read Only" flag status.

Is this expected behavior?

georgweiss, 2020-02-28:
Yes it is.  However, I will change this (soon) such that the flag can be modified. 
On top of that I also intend to make it possible to rename a PV in the save set. This however has some implications: a PV name must be unique and is stored in the database as a unique record. Changing a PV name would then change it for all save sets and their snapshots.
I also intend to make it possible to export a list of PVs from the channel finder into a new or existing save set.

geniejhang, 2020-02-28:
Okay. Now I understand the behaviors. Thanks.
In addition to that, I implemented half of the feature in your last sentence.
https://github.com/geniejhang/phoebus/commit/7019fa5f5fc718855bfc2be37e8a6ec615deda22

Creating a saveset from a selection of PV names via context menu.

georgweiss, 2020-02-28:
Cool, I will try it out as soon as possible.

geniejhang, 2020-03-03:
@georgweiss Updated for fixing a stuipid bug. https://github.com/geniejhang/phoebus

geniejhang, 2020-03-05:
Just for your information, I tried to add ~5000 PVs using feature I added.
It took ~7s raising time out error with the default time out value (1000ms).

georgweiss, 2020-03-06:
Thanks for the heads-up.
Actually, a use case for 5000 PVs may in practice not be a very common one. The  save-and-restore is not intended for backup, but instead facilitate activities like calibration and fine tuning. Listing 5000 PVs in a snapshot will not be very usable, UI wise. It would also put quite some load on the host as 5000 PVs must be monitored and updated in a table.
Anyhow, the UI can be configured to use a longer timeout when accessing the back-end service, see save_and_restore_preferences.properties. What should be added to the UI is a progress indicator, I will add this on my backlog.

closed 2020-03-06 by georgweiss