ControlSystemStudio/phoebus #1180: Embedded display default value not persisted

georgweiss, 2020-03-02:
If an embedded display is configured to have a height of 300 the value is not persisted in the bob file. When launching (or reloading in edit mode) the OPI the embedded display gets a different size, which is incorrect with respect to the design of the OPI and the embedded display.

What is the reason to not persist a height value only because it coincides with the default property value?

The "culprit" code is in ModelWriter#writeWidgetProperties, line 153.

kasemir, 2020-03-02:
Default values are generally not persisted since they are, well, the default, and the widget gets that value without having to spell it out in the file.
Can you attach an example pair of main & embedded displays where this causes a problem?

georgweiss, 2020-03-03:
[AreaDetector_20200211.zip](https://github.com/ControlSystemStudio/phoebus/files/4279946/AreaDetector_20200211.zip)
Open ADSpinnaker/bob/autoconvert/ADSpinnaker_v01.bob and observe the Trigger embedded display (right middle). It supposed to be 300 in height, but when run it's set to 200 as this is the height set in EmbeddedDisplayRepresentation:
`@Override
    protected void defineProperties(final List<WidgetProperty<?>> properties)
    {
        ...
        // Initial size
        propWidth().setValue(300);
        propHeight().setValue(200);
    }`

The bob file is saved with Resize Behavior=No resize, so I guess this explains the observed behavior. If I change to other resize behavior I get the wanted height (300). Nevertheless I find it odd to not honor/persist the desired height setting only because it happens to be the same as the default (which is overridden anyway in defineProperties).

closed 2020-03-06 by georgweiss