ControlSystemStudio/phoebus #1182: Online help no longer includes plain html from doc sources

kasemir, 2020-03-02:
The documentation contains some plain html, like the javadoc for display builder scripts.

https://github.com/ControlSystemStudio/phoebus/blob/master/docs/README.md still describes that it's supported:

```
# Build the Javadoc, i.e. html files to be included in the manual
( cd phoebus/app/display/editor; ant -f javadoc.xml clean all )

# Building the documentation
( cd phoebus/doc; make clean html )
```

.. but in the generated manual, those links don't work.
It's not just that the path to html that's copied in changed, the content seems simply missing.

Same when calling `mvn verify -P sphinx -N`

kasemir, 2020-03-02:
#1141

shroffk, 2020-03-02:
Updated the conf.py to copy generated html

The generated html is copied into 
build/html/raw_html/<module>/<path>...

e.g.
[image]

There is a change in the way the `<a href="html/generated/index.html">Java Doc for scripts</a>` is resolved.

Previously it resolved to to:
`phoebus/app/display/editor/doc/html/generated/index.html`

now it resolves to:
`phoebus/docs/build/html/html/generated/index.html`

The reason behind this is because the use of applications_links.rst instead of the applications.rst

shroffk, 2020-03-02:
While the old way in which the applications.rst was used meant the git would always see that file as modified which each run. I still feel like we might want to revert to the old way of modifying the application.rst directly to simply the linking of raw html

kasemir, 2020-03-02:
Your pick. All I want is to have online help.

shroffk, 2020-03-02:
Noted. Thank you

kasemir, 2020-03-02:
> Previously it resolved to to:
> phoebus/app/display/editor/doc/html/generated/index.html
> now it resolves to:
> phoebus/docs/build/html/html/generated/index.html

What if both `display/editor/doc` had some plain html, and `other/whatever/doc` also has some plain html. Would they both be copied into the same `docs/build/html/html`?

shroffk, 2020-03-02:
>  Would they both be copied into the same docs/build/html/html?

No this would surely not work. The static html has to go to a module specific location so that multiple modules can contribute their own html and other static content.

shroffk, 2020-03-02:
The last commit does the following
Copy the doc folder mainiting the same structure as the phoebus src.
Creates the application.rst and services.rst 
Copies exiting html to app specific directories e.g. `phoebus/app/display/editor/doc/html/generated/index.html`

closed 2020-03-09 by kasemir