ControlSystemStudio/phoebus #1186: Modified display runs in new tab

georgweiss, 2020-03-03:
The following behaviour has triggered complaints:
1) Launch display from editor.
2) Go back to editor to modify and save display.
3) Launch again from editor -> new tab is created.

One could consider reusing the tab of the running display, perhaps controlled by a preference.
If no-one objects I can do it provided tabs can be uniquely identified through the file name. For the sake of consistency, the same strategy should apply if the display is running in a detached window.

kasemir, 2020-03-09:
> provided tabs can be uniquely identified through the file name.

They're identified by their 'resource' URL, which includes the file name and macros.

georgweiss, 2020-03-09:
OK, so I will go ahead and implement this. Any reason to condition the changed behaviour through a preference setting?

kasemir, 2020-03-09:
In principle, that's already implemented. The display runtime looks for existing runtimes with the same input. Check in which way the resource URL changes to make it look like a different filename and macros.

georgweiss, 2020-03-09:
OK, will do.

kasemir, 2020-03-09:
Check `DisplayRuntimeApplication.create(URI)`. It tries to locate an existing instance with the same 'input' so it can raise that instead of  creating another copy.

georgweiss, 2020-03-11:
Turns out the observed behaviour is caused by inconsistent handling of spaces in file names. In DockStage.getDockItemWithInput() the if statement
`if (input.equals(item.getInput()) &&
                            item.getApplication().getAppDescriptor().getName().equals(application_name))
                            return item;`
will compare different representations of the `input` URI and value returned by `item.getItem()`. In the case of spaces it will be `+` and `%20` respectively.

I'd prefer to hand this over to anyone with more experience with the underlying classes.

kasemir, 2020-03-11:
@DominicOram Dominic, you recently worked on `ModelResourceUtil.normalize` et al. in #1140
Want to give this a try?

georgweiss, 2020-03-11:
Is `ModelResourceUtil.normalize` the right place? I'm asking since `DisplayInfo.toURI` also manipulates the path (replaces space with +)...

georgweiss, 2020-03-12:
Any updates on this?

DominicOram, 2020-03-12:
I haven't looked at this and whilst I would like to be able to do some work on it it's unlikely I will have the time in the foreseeable future.

georgweiss, 2020-03-13:
@kasemir and @DominicOram , this is clearly a bug in the core framework, most probably due to conversions between URIs and file paths. Using spaces in file names is not an uncommon use case, so I need to insist that this would be better corrected by developers with more experience in affected APIs.

shroffk, 2020-03-13:
Was this fixed?
I can't seem to reproduce this error.

I am creating a file with a space in the file name... "test file.opi"

When I open it with editor, it reuses the same tab that is currently open. In the DockStage both the URI's have the spaces parsed correctly (i.e. %20)

shroffk, 2020-03-13:
I think I was able to reproduce it... the issue arises only with the runtime and not with the editor.

georgweiss, 2020-03-13:
Correct, if a display is launched repeatedly from the editor, new tabs are created. Your observation confirms my assumption: edit and runtime handle URIs and file names containing blanks inconsistently.

shroffk, 2020-03-13:
Ok, 
So I have what I think is a rather ugly solution.
I have tried to hack the existing windows os tweaks to try to conform to the following.
https://en.wikipedia.org/wiki/File_URI_scheme

I initially tried to use Paths & URI to parse and handle the resources uri being passed around, while I was able to get the tabs to be reused... I had trouble with various unit tests and the way query and path parameters were extracted. I abandoned the effort to use File, Path, and URI and instead adopted the above hack.

I know I am asking to break working code... but it would be nice to try to use java packages as much as possible over string concatenation.

@georgweiss please test this branch and let me know what other issues are there... since I have very low confidence in the patch I have submitted.

georgweiss, 2020-03-16:
@shroffk , your patch solves the issue with blanks in paths (verified on Mac and Windows). However, a display with macros will still always open new tabs/windows. The comparison in `DockStage.getDockItemWithInput()` fails because the item.getInput() will return URI where marcos are present as request parameters. But this is maybe a different issue?

shroffk, 2020-03-16:
Hmm.. this should be using the URI equals.

I can't seem to reproduce this error.
https://drive.google.com/drive/folders/1fgeqVcR-h8l9ox8uZy3ukWpuDXVcQqc7?usp=sharing

there are 2 files...
one which opens a second one with macros...if you press the button it opens the same opi every time.

shroffk, 2020-03-16:
I also tried Macro values with spaces in them...

kasemir, 2020-03-16:
One quirk that I sometimes see is that the default macros are in the runtime link (https://github.com/ControlSystemStudio/phoebus/blob/master/app/display/model/src/main/resources/display_model_preferences.properties#L35, EXAMPLE_MACRO=..., TEST=...).
A link without any macros and one with the default macros are considered different, which I think is correct. Any change in macros could result in a different look, so the runtime needs to keep those separate.

georgweiss, 2020-03-16:
@shroffk , your test fails if you add a macro to "test space.bob", on "top level", i.e. not for the button action.

georgweiss, 2020-03-16:
That's fine with me, I will close it and open a new.

closed 2020-03-16 by georgweiss