ControlSystemStudio/phoebus #1195: core-ui compilation error

hanlet, 2020-03-10:
Hi, I just cloned the latest and greatest version of phoebus, but am having
an issue with compilation using maven.  I'm behind a firewall, so I use
a socks proxy:
> proxychains mvn clean install -DskipTests

INFO] ------------------------------------------------------------------------
[INFO] Building core-ui 4.6.2-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.4.1:clean (default-clean) @ core-ui ---
[INFO] Deleting /usr/local/phoebus/core/ui/target
[INFO] 
[INFO] --- maven-resources-plugin:2.5:resources (default-resources) @ core-ui ---
[debug] execute contextualize
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 71 resources
[INFO] 
[INFO] --- maven-compiler-plugin:3.6.2:compile (default-compile) @ core-ui ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 113 source files to /usr/local/phoebus/core/ui/target/classes
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] /usr/local/phoebus/core/ui/src/main/java/org/phoebus/ui/examples/ExampleInstaller.java:[145,54] reference to newFileSystem is ambiguous
  both method newFileSystem(java.nio.file.Path,java.lang.ClassLoader) in java.nio.file.FileSystems and method newFileSystem(java.nio.file.Path,java.util.Map<java.lang.String,?>) in java.nio.file.FileSystems match
[INFO] 1 error
[INFO] -------------------------------------------------------------
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] phoebus (parent) .................................. SUCCESS [0.514s]
[INFO] dependencies ...................................... SUCCESS [0.046s]
[INFO] install-jars ...................................... SUCCESS [0.503s]
[INFO] phoebus-target .................................... SUCCESS [4.544s]
[INFO] core .............................................. SUCCESS [0.035s]
[INFO] core-vtype ........................................ SUCCESS [1.758s]
[INFO] core-formula ...................................... SUCCESS [1.037s]
[INFO] core-framework .................................... SUCCESS [1.571s]
[INFO] core-types ........................................ SUCCESS [0.090s]
[INFO] core-util ......................................... SUCCESS [0.691s]
[INFO] core-pva .......................................... SUCCESS [1.612s]
[INFO] core-pv ........................................... SUCCESS [0.901s]
[INFO] core-security ..................................... SUCCESS [0.200s]
[INFO] core-ui ........................................... FAILURE [1.093s]
.
.
.
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 15.038s
[INFO] Finished at: Tue Mar 10 15:34:38 CDT 2020
[INFO] Final Memory: 39M/144M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.6.2:compile (default-compile) on project core-ui: Compilation failure
[ERROR] /usr/local/phoebus/core/ui/src/main/java/org/phoebus/ui/examples/ExampleInstaller.java:[145,54] reference to newFileSystem is ambiguous
[ERROR] both method newFileSystem(java.nio.file.Path,java.lang.ClassLoader) in java.nio.file.FileSystems and method newFileSystem(java.nio.file.Path,java.util.Map<java.lang.String,?>) in java.nio.file.FileSystems match
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :core-ui

shroffk, 2020-03-10:
Strange java.nio. error... I don't think your proxy should be an issue.
What version of java are you using.

kasemir, 2020-03-10:
Must be a very new version where there is indeed another variant of `newFileSystem`.

Try changing ExampleInstaller line 145 to this:
```
final FileSystem fs = FileSystems.newFileSystem(jar, (ClassLoader) null);
```

kasemir, 2020-03-10:
So far, there was only one `newFileSystem` method that took a `Path`, but your error shows two variants.

hanlet, 2020-03-10:
openjdk version "13.0.2" 2020-01-14
OpenJDK Runtime Environment (build 13.0.2+8)
OpenJDK 64-Bit Server VM (build 13.0.2+8, mixed mode, sharing)

hanlet, 2020-03-10:
Thank you, that worked.  I've a new issue now; but I'll close this one 
and start a new one.

On 3/10/20 3:55 PM, Kay Kasemir wrote:
> |final FileSystem fs = FileSystems.newFileSystem(jar, (ClassLoader) null);|

-- 
Pierrick Hanlet
+1-312-687-4980       US
+44 (0)79-48-86-01-97 UK

"Whether you think you can or think you can't, either way, you are correct" -- Henry Ford

closed 2020-03-10 by hanlet