ControlSystemStudio/phoebus #1197: save and restore build error

hanlet, 2020-03-10:
Using openjdk version "13.0.2" 2020-01-14

[INFO] service-save-and-restore .......................... FAILURE [18.563s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 2:23.737s
[INFO] Finished at: Tue Mar 10 16:05:24 CDT 2020
[INFO] Final Memory: 80M/287M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-javadoc-plugin:3.0.1:jar (attach-javadocs) on project service-save-and-restore: MavenReportException: Error while generating Javadoc:
[ERROR] Exit code: 1 - /usr/local/phoebus/services/save-and-restore/src/main/java/org/phoebus/service/saveandrestore/package-info.java:4: error: unexpected heading used: <H1>, compared to implicit preceding heading: <H1>
[ERROR] * <h1>Client aspects</h1>
[ERROR] ^
[ERROR] /usr/local/phoebus/services/save-and-restore/src/main/java/org/phoebus/service/saveandrestore/package-info.java:10: error: unexpected heading used: <H1>, compared to implicit preceding heading: <H1>
[ERROR] * <h1>Key concepts</h1>
[ERROR] ^
[ERROR] /usr/local/phoebus/services/save-and-restore/src/main/java/org/phoebus/service/saveandrestore/package-info.java:62: error: unexpected heading used: <H1>, compared to implicit preceding heading: <H1>
[ERROR] * <h1>Database implementation</h1>
[ERROR] ^
[ERROR] /usr/local/phoebus/services/save-and-restore/src/main/java/org/phoebus/service/saveandrestore/package-info.java:74: error: unexpected heading used: <H1>, compared to implicit preceding heading: <H1>
[ERROR] * <h1>Client work flow example</h1>
[ERROR] ^
[ERROR] 
[ERROR] Command line was: /usr/local/jdk-13.0.2/bin/javadoc @options @packages
[ERROR] 
[ERROR] Refer to the generated Javadoc files in '/usr/local/phoebus/services/save-and-restore/target/apidocs' dir.
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :service-save-and-restore

georgweiss, 2020-03-11:
Fixed.

hanlet, 2020-03-11:
Thank you.  It's working now.

closed 2020-03-11 by hanlet