ControlSystemStudio/phoebus #1200: Suppressing PV writes from spinner if value is outside limits.

georgweiss, 2020-03-11:
Users are requesting the a change to the spinner widget like so:
1) User unintentionally enters a value that falls outside the configured limits.
2) The spinner widget should then prompt the user that the entered value is "invalid", and submit the value. The current behaviour is to write the maximum or minimum value to the PV.

If this change is too invasive, we can add a setting or - even better? - add a property to the spinner widget.

kasemir, 2020-03-11:
Prompts should be a rare exception. Can't block the UI with modal dialogs or clutter the screen with non-modal dialogs.

With a text widget, the user can enter anything and the IOC then decides what to do with the value: Ignore? Clamp to limit? Use?
With a slider, you get a fixed control range and can't go beyond.
The spinner up/down buttons similarly enforce the control range. Since the widget thus already has a 'clamping' behavior, it makes sense that it also clamps the manually entered value.

If you want to change that for ESS, do it as a preference setting that defaults to 'clamp', the current behavior. I don't want a property which is then set to this in one display and that in another, so end users never know how a spinner will behave. This needs to be consistent for each site.

georgweiss, 2020-03-11:
I'm not a fan of dialogs either, but if we support clamping we need to tell the user that the value was not written, and why.
Unlike the spinner, the text widget cannot be configured with limits. The issue our users have with the spinner is entering values in the text field, which sometimes turns out wrong. Depending on the use case, writing the maximum value automatically may have consequences that take time to rectify.

I'd prefer if this could be done with a script, but I suppose the PV is written before the script is evaluated, right?