ControlSystemStudio/phoebus #1203: restore configuration in data browser

hanlet, 2020-03-11:
I've noticed that in the newest version of Phoebus, that many of the data browser settings, e.g. fonts
are now restored when re-instantiating a new Phoebus browser.  Thanks for that.
However, the horizontal axis name is still not saved.

kasemir, 2020-03-11:
There is no horizontal axis name. The time axis has no configurable label.
Do you mean the "Title" on top of the plot? That's saved when I try it.

hanlet, 2020-03-11:
Hi Kay,
Please see attached; the window opens when clicking on the Plot Options 
button.
Cheers,
Pierrick

kasemir, 2020-03-12:
Sorry, oversight. These are runtime settings which unfortunately include for example the option of hiding the time axis, which is clearly a bad idea. Will remove them.