ControlSystemStudio/phoebus #1207: How to specify directories with related displays?

prjemian, 2020-03-13:
Looking for something like MEDM's `EPICS_DISPLAY_PATH` setting that gives a directory or directories that contain display files that may be called.

Basically, how to use something like synApps with Phoebus?

prjemian, 2020-03-13:
In the context of (within an `.opi` file):

```
  <widget typeId="org.csstudio.opibuilder.widgets.MenuButton" version="1.0.0">
    <actions hook="false" hook_all="false">
      <action type="OPEN_DISPLAY">
        <path>topMotors8.opi</path>
        <macros>
          <include_parent_macros>true</include_parent_macros>
          <P>xxx:</P>
          <M1>m1</M1>
          <M2>m2</M2>
          <M3>m3</M3>
          <M4>m4</M4>
          <M5>m5</M5>
          <M6>m6</M6>
          <M7>m7</M7>
          <M8>m8</M8>
        </macros>
        <mode>1</mode>
        <description>1-8</description>
      </action>
      <!-- more such actions -->
  </widget>
```

the file `topMotors8.opi` will be found in a different directory than the file that contains this action.

kasemir, 2020-03-13:
There is no display search path. On the SNS accelerator, we use EDM which has a search path similar to MEDM, and the result is that you never know which file you'll get. The display builder was thus designed to only support absolute or - and that is the preferred setup - relative paths. That way, you know exactly which file will be opened.

For translating EDM files, the EDM auto-converter supports an EDM search path, so it can find all EDM files and then convert them into one target folder. The MEDM converter currently doesn't have such a feature, so you need to convert all folders with MEDM files into one, or adjust the links.

https://github.com/ControlSystemStudio/phoebus/blob/master/app/display/convert-medm/doc/index.rst#converting-edm-edl-displays

I understand that this is a change for MEDM or EDM users, but then web pages are another great example of a system that is also limited to absolute or relative links without a search path, and it seems to suffice for the rest of the world.

prjemian, 2020-03-13:
Thanks for the quick answer.  Now, we have a record of this fundamental design decision.  It would help to add something about this as a note in the documentation, to help others who transition from display tools that provide a search path.

closed 2020-03-13 by prjemian

kasemir, 2020-03-13:
OK, added to https://github.com/ControlSystemStudio/phoebus/wiki/Display-Builder-Compatibility#paths