ControlSystemStudio/phoebus #1210: Displays with top level macros always open in new tab/window.

georgweiss, 2020-03-16:
When DockStage looks for open displays the below comparison (line 291) fails if a display has top level macros:
`if (input.equals(item.getInput()) &&
                            item.getApplication().getAppDescriptor().getName().equals(application_name))
                            return item;`
Root cause us that input is the URI without macros, while item.getInput() return the same URI, but augmented with macros.

kasemir, 2020-03-17:
See #928

georgweiss, 2020-03-17:
Not sure #928 applies. My point is that in DockStage an URI with macros is compared to an URI without macros, even though the two URIs represent the same OPI file.

kasemir, 2020-03-17:
A display is identified by the file name and macros.
You want to open file "x".
A tab replies with "I am file x with macros M=V".
On one hand, they're different.
On the other hand, "M=V" could be the default macros, they they would end up being the same.

One fix would be that an attempt to open file "x" gets extended with the default macros, so you'd compare "I want to open file x with macros M=V" against "I am file x with macros M=V".

georgweiss, 2020-03-17:
I do not think that is the issue.
The method DockStage#getDockItemWithInput is called with an URI that is constructed from a file path only, so macro information is not available. This URI is compared to the input field of an DockItemWithInput, where macros are appended (not default macros, though). To me this is inconsistent and confusing: depending on context the input URI may come with or without macros.
Attached screen shot from a debugging session  demonstrates my point.
[image]

georgweiss, 2020-03-18:
I'd would appreciate some attention for this issue. I may be wrong, but in my view the root cause is that the implementation attempts to support both URIs and file paths depending on how a display is loaded. The data objects (e.g. DisplayInfo) when edit and runtime applications are started are not created in a consistent manner between those cases. Ultimately this results in a comparison in DockStage that will never be meaningful.
Also detected that a file URI specified as the quoted value for the resource application argument does not work. The quotes would be needed to support space characters.

georgweiss, 2020-03-19:
Will not fix this. Closing due to lack of input.

closed 2020-03-19 by georgweiss