ControlSystemStudio/phoebus #1211: bob displays not connecting

hanlet, 2020-03-16:
I'm assuming that I created this problem, but I've been unsuccessful at resolving it.
2020-03-16 13:13:41 WARNING [com.cosylab.epics.caj.impl.BroadcastTransport] Failed to sent a datagram to:1131.225.120.227:6716
java.nio.channels.UnresolvedAddressException

I changed my settings.ini file, but my addresses in org.phoebus.pv.ca/addr_list don't have
the additional "1" in the IP addresses.

I'm using the latest phoebus from github and jdk-13.0.2 on a linux SL 64bit pc.

shroffk, 2020-03-16:
Are you sure there is no typo in the settings.ini file? or a duplicate entry for the address list?

The settings.ini are supposed to override other stored preferences

You can remove the following folder ```~/.phoebus/.userPrefs/org/phoebus/pv/ca``` and try again.
This is the folder which caches the preferences.

kasemir, 2020-03-16:
Also check Help, About, Show Details, Preference Settings what it lists for org.phoebus.pv.ca/addr_list.
Finally, the CAJ lib actually looks at environment variables and system properties.
We set the system properties from the phoebus/java preference settings, and that should replace environment variables, but just to make sure check if `set | fgrep 1131` includes that IP address in some environment variable.

kasemir, 2020-03-16:
When you start phoebus, there should be log messages which show you the settings files that are loaded:
```
Loading settings from ...settings.ini
```
Maybe it's not loading the settings that you expect?

hanlet, 2020-03-16:
Hi gents,
That was the problem; I'm not quite certain what happened.  In the end, 
I deleted the
entire ~/.phoebus directory.  All is well now.
Thank you,
Pierrick

closed 2020-03-16 by kasemir