ControlSystemStudio/phoebus #1225: Alarm view showing wrong connection status

georgweiss, 2020-03-20:
Alarm information for PVs that are offline when the alarm server starts will show wrong alarm severity and status (UNDEFINED/Disconnected) in the alarm views when the PVs come back online. The "PV severity" is however correct (MAJOR, MINOR).
Same thing happens if the PVs goes offline during normal alarm server operation.

kasemir, 2020-03-20:
What's wrong?
Offline PVs are UNDEFINED/Disconnected. If the PV later connects and the 'current' state is MAJOR or MINOR, the latched alarm severity remains UNDEFINED/Disconnected.
If you acknowledge the alarm, it turns into UNDEFINED_ACK.
If the PV finally recovers, it turns into OK.
That's the design, that's what's tested in https://github.com/ControlSystemStudio/phoebus/blob/master/services/alarm-server/src/test/java/org/phoebus/applications/alarm/server/AlarmLogicUnitTest.java#L292, no change from the original design when it was implemented for Eclipse.

georgweiss, 2020-03-20:
OK, I was not aware of this application logic. And if this is what users are familiar with we should not change it.

closed 2020-03-20 by georgweiss