ControlSystemStudio/phoebus #1226: Save and restore error

MaticPogacnik, 2020-03-20:
Hi,
I am trying to start save-and-restore service version 4.6.1, but it looks like connections to Database (MySQL 8.0.17) fails, with the error bellow (any tips why this happens).
I start the service with:

```
java -Ddbengine=mysql -Dhostname=localhost -Dspring.datasource.username=masar -Dspring.datasource.password=masar -Dspring.datasource.jdbcUrl=jdbc:mysql://localhost:3306/masar?useSSL=false -jar target/service-save-and-restore-4.6.1.jar
```
Thank you.

```

2020-03-20 15:12:21.073 [main] INFO  org.phoebus.service.saveandrestore.application.Application - Starting Application on localhost.localdomain with PID 20930 (/home/devman/git/phoebus/services/save-and-restore/target/service-save-and-restore-4.6.1.jar started by devman in /home/devman/git/phoebus/services/save-and-restore)
2020-03-20 15:12:21.093 [main] INFO  org.phoebus.service.saveandrestore.application.Application - No active profile set, falling back to default profiles: default
2020-03-20 15:12:23.554 [main] INFO  org.springframework.context.support.PostProcessorRegistrationDelegate$BeanPostProcessorChecker - Bean 'org.springframework.transaction.annotation.ProxyTransactionManagementConfiguration' of type [org.springframework.transaction.annotation.ProxyTransactionManagementConfiguration$$EnhancerBySpringCGLIB$$9b75c9df] is not eligible for getting processed by all BeanPostProcessors (for example: not eligible for auto-proxying)
2020-03-20 15:12:24.055 [main] INFO  org.springframework.boot.web.embedded.tomcat.TomcatWebServer - Tomcat initialized with port(s): 8080 (http)
2020-03-20 15:12:24.131 [main] INFO  org.apache.coyote.http11.Http11NioProtocol - Initializing ProtocolHandler ["http-nio-8080"]
2020-03-20 15:12:24.164 [main] INFO  org.apache.catalina.core.StandardService - Starting service [Tomcat]
2020-03-20 15:12:24.165 [main] INFO  org.apache.catalina.core.StandardEngine - Starting Servlet engine: [Apache Tomcat/9.0.19]
2020-03-20 15:12:24.341 [main] INFO  org.apache.catalina.core.ContainerBase.[Tomcat].[localhost].[/] - Initializing Spring embedded WebApplicationContext
2020-03-20 15:12:24.341 [main] INFO  org.springframework.web.context.ContextLoader - Root WebApplicationContext: initialization completed in 3030 ms
2020-03-20 15:12:25.363 [main] INFO  org.flywaydb.core.internal.license.VersionPrinter - Flyway Community Edition 5.2.4 by Boxfuse
2020-03-20 15:12:25.385 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Starting...
2020-03-20 15:12:25.771 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Start completed.
2020-03-20 15:12:25.780 [main] INFO  org.flywaydb.core.internal.database.DatabaseFactory - Database: jdbc:mysql://localhost:3306/masar (MySQL 8.0)
2020-03-20 15:12:26.046 [main] WARN  org.springframework.boot.web.servlet.context.AnnotationConfigServletWebServerApplicationContext - Exception encountered during context initialization - cancelling refresh attempt: org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'configurationController': Unsatisfied dependency expressed through field 'services'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'services': Unsatisfied dependency expressed through field 'nodeDAO'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'configDAO': Unsatisfied dependency expressed through field 'jdbcTemplate'; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'flywayInitializer' defined in class path resource [org/springframework/boot/autoconfigure/flyway/FlywayAutoConfiguration$FlywayConfiguration.class]: Invocation of init method failed; nested exception is org.flywaydb.core.api.FlywayException: Validate failed: Detected failed migration to version 8.0 (snapshot as node)
2020-03-20 15:12:26.047 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Shutdown initiated...
2020-03-20 15:12:26.066 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Shutdown completed.
2020-03-20 15:12:26.075 [main] INFO  org.apache.catalina.core.StandardService - Stopping service [Tomcat]
2020-03-20 15:12:26.182 [main] INFO  org.springframework.boot.autoconfigure.logging.ConditionEvaluationReportLoggingListener - 

Error starting ApplicationContext. To display the conditions report re-run your application with 'debug' enabled.
2020-03-20 15:12:26.184 [main] ERROR org.springframework.boot.SpringApplication - Application run failed
org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'configurationController': Unsatisfied dependency expressed through field 'services'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'services': Unsatisfied dependency expressed through field 'nodeDAO'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'configDAO': Unsatisfied dependency expressed through field 'jdbcTemplate'; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'flywayInitializer' defined in class path resource [org/springframework/boot/autoconfigure/flyway/FlywayAutoConfiguration$FlywayConfiguration.class]: Invocation of init method failed; nested exception is org.flywaydb.core.api.FlywayException: Validate failed: Detected failed migration to version 8.0 (snapshot as node)
        at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.inject(AutowiredAnnotationBeanPostProcessor.java:596) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.annotation.InjectionMetadata.inject(InjectionMetadata.java:90) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor.postProcessProperties(AutowiredAnnotationBeanPostProcessor.java:374) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.populateBean(AbstractAutowireCapableBeanFactory.java:1411) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:592) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:515) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:320) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:222) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:318) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:199) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.preInstantiateSingletons(DefaultListableBeanFactory.java:843) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.context.support.AbstractApplicationContext.finishBeanFactoryInitialization(AbstractApplicationContext.java:877) ~[spring-context-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:549) ~[spring-context-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.refresh(ServletWebServerApplicationContext.java:142) ~[spring-boot-2.1.5.RELEASE.jar!/:2.1.5.RELEASE]
        at org.springframework.boot.SpringApplication.refresh(SpringApplication.java:775) [spring-boot-2.1.5.RELEASE.jar!/:2.1.5.RELEASE]
        at org.springframework.boot.SpringApplication.refreshContext(SpringApplication.java:397) [spring-boot-2.1.5.RELEASE.jar!/:2.1.5.RELEASE]
[... cut ...]
        at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor.postProcessProperties(AutowiredAnnotationBeanPostProcessor.java:374) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.populateBean(AbstractAutowireCapableBeanFactory.java:1411) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:592) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactor
```

georgweiss, 2020-03-20:
Apparently a previous startup failed on the Flyway migration script for version 8. I have heard similar reports that turned out to be related to the use of database clusters. Are you using a single host for Mysql, or a clustered setup?

MaticPogacnik, 2020-03-20:
Thanks you for the fast response. I am running single instance (on my laptop).

There seems to be invalid syntax in the V8 sql script (with which version of mysql this was tested?)

```
Caused by: org.flywaydb.core.internal.sqlscript.FlywaySqlScriptException:
Migration V8_0__snapshot_as_node.sql failed
-------------------------------------------
SQL State  : HY000
Error Code : 1419
Message    : You do not have the SUPER privilege and binary logging is enabled (you *might* want to use the less safe log_bin_trust_                                                           function_creators variable)
Location   : db/migration/mysql/V8_0__snapshot_as_node.sql (/home/devman/nusano/phoebus/services/save-and-restore/file:/home/devman/                                                           nusano/phoebus/services/save-and-restore/target/service-save-and-restore-4.6.1.jar!/BOOT-INF/classes!/db/migration/mysql/V8_0__snaps                                                           hot_as_node.sql)
Line       : 9
Statement  : CREATE TRIGGER trigger_id_node BEFORE INSERT ON node
    FOR EACH ROW BEGIN
        DECLARE incremented_number BIGINT UNSIGNED;
        UPDATE global_sequence SET number = number + 1;
        SELECT number INTO incremented_number
            FROM global_sequence;
        SET new.id := incremented_number;
    END

        at org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor.handleException(DefaultSqlScriptExecutor.java:253) ~[flyway                                                           -core-5.2.4.jar!/:?]

```

before running this I droped the `masar` database and created it again.

georgweiss, 2020-03-20:
I have verified this on Mysql 8.0.19, so I would not suspect the syntax to be wrong. To me it looks like a permission issue. This might be of some assistance: http://wiki.ispirer.com/sqlways/troubleshooting-guide/mysql/import/binary-logging

MaticPogacnik, 2020-03-21:
Thank you for the useful tips, here are the binary logging setting and permissions settings:

```
mysql> show global variables like 'log_bin_trust_function_creators';
+---------------------------------+-------+
| Variable_name                   | Value |
+---------------------------------+-------+
| log_bin_trust_function_creators | OFF   |
+---------------------------------+-------+
1 row in set (0.00 sec)

mysql> show grants for masar;
+--------------------------------------------------+
| Grants for masar@%                               |
+--------------------------------------------------+
| GRANT USAGE ON *.* TO `masar`@`%`                |
| GRANT ALL PRIVILEGES ON `masar`.* TO `masar`@`%` |
+--------------------------------------------------+
2 rows in set (0.00 sec)

mysql>

```

MaticPogacnik, 2020-03-21:
If I give the user super permissions flag with:

```
mysql> UPDATE mysql.user SET Super_Priv='Y' WHERE user='masar' AND host='%';
Query OK, 1 row affected (0.01 sec)
Rows matched: 1  Changed: 1  Warnings: 0
```

It kinda works, but i get two warrnings:

```
...
2020-03-21 16:21:23.970 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `masar` to version 8.0 - snapshot as node
2020-03-21 16:21:24.044 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Trigger does not exist (SQL State: HY000 - Error Code: 1360)
2020-03-21 16:21:24.073 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Trigger does not exist (SQL State: HY000 - Error Code: 1360)
2020-03-21 16:21:24.440 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `masar` to version 9.0 - add golden to snapshot
2020-03-21 16:21:24.507 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `masar` to version 10.0 - add unique id to node
...
```

georgweiss, 2020-03-21:
I'm seeing the same. It's safe to ignore the warning.

MaticPogacnik, 2020-03-21:
Great,

Thank you for you assistance.

closed 2020-03-21 by MaticPogacnik