ControlSystemStudio/phoebus #1231: Display Navigation

shroffk, 2020-03-23:
At the developers meeting it was discussed that having a tool or application that created a list/graph/tree of all the opi files linked to a particular opi might be useful.

This was aimed at helping with the migration from boy to display builder.

I felt like such an app/tool would be useful in normal circumstances to, when developing and managing screens it is always useful to understand the links from that screen.

I have created an application called display navigation view which parses a given opi screen and produces 
1. a unique list of all linked files
2. a tree view of all linked files.

This is the first iteration and we can all more feature as we need them.

[image]

[image]

shroffk, 2020-03-23:
@tanviash let me know if this work for you

mark0n, 2020-03-23:
I like the idea of generating a tree that you can navigate (hide subtrees). I'm wondering how you are visualizing if your graph is cyclic (remember we want to find cycles)? Have you tried throwing your data into [GraphViz](https://graphviz.org) to get a graphical representation? There seems to be a [Java interface](https://github.com/nidi3/graphviz-java) as well.

shroffk, 2020-06-04:
@mark0n @tanviash while the graph is fun the current app...though not pretty... fulfils the core requirements.
Can we close this ticket and open a new one for the graphing

shroffk, 2020-06-04:
oops closed by mistake... you can reopen if necessary

closed 2020-06-04 by shroffk