ControlSystemStudio/phoebus #1233: OpenJDK 14

kasemir, 2020-03-24:
The current release of openJDK is 14, http://openjdk.java.net/projects/jdk/14/

It it includes a `jpackage` preview which could help with wrapping the launcher as a native exe/app/.. #67

Also includes a helpful NullPointerException update (needs to be enabled via `-XX:+ShowCodeDetailsInExceptionMessages`).

First tests indicate that CS-Studio/Phoebus compiles and runs fine w/ JDK 14. Binaries compiled w/ previous version also run OK with JRE 14.

georgweiss, 2020-03-31:
I have spent a short time on this and it looks fairly promising. Customizing an installer in terms of look and feel may not be possible.
On Mac you may create a "app" package to put directly in your /Applications folder, or a dmg that wraps the same thing.
On Windows I've only tried the msi option, which results in an easy-to-use installer.
It remains to be investigated how to customize launcher with JVM options and application arguments.
Have not attempted Linux packaging.

kasemir, 2020-05-29:
Looked into using `jpackage` on Mac, Windows and Linux.

It basically works, creating a runnable application with bundled JRE.
Figured out how to add online help, and a second `jpackage` pass can then create an installer as common for the OS (dmg, exe, rpm).

But it needs to be performed on each OS, i.e. cannot package for all platforms on a Linux host, and the binary launcher won't support online update.
Details are on https://github.com/ControlSystemStudio/phoebus/wiki/Packaging

For the SNS product I'm staying with building and bundling for all platforms on Linux (Hudson/jenkins), bundling a JRE into the Win & Mac version and using a launcher script that supports online update. Unfortunately, the result is a ZIP and not an installer, but right now that seems like the best compromise.

Will check again with future JDK updates.

closed 2020-05-29 by kasemir