ControlSystemStudio/phoebus #1235: Save and Restore: PVs disconnected

dsoklic, 2020-03-25:
In the Save and Restore application sometimes the PVs don't want to connect. I can however always access the PVs in other places, like the PV Table. The error seems to occur randomly. I can run Phoebus and everything works fine and then at the next run without changing anything the PVs can not be accessed, but only from Save and Restore.

I have attached a screenshot where this can be seen and also a screenshot of the PV table where the same PV does connect.

[image]
[image]

georgweiss, 2020-03-25:
Please attach output of terminal.

dsoklic, 2020-03-25:
Thank you for your quick reply. Here is the entirety of today's log:

[phoebus_save_and_restore.log](https://github.com/ControlSystemStudio/phoebus/files/4380737/phoebus_save_and_restore.log)

georgweiss, 2020-03-25:
I'm sorry, need to clarify: need the output from the CS Studio terminal. The service does not connect to PVs.

dsoklic, 2020-03-25:
Sorry about that. Here is the output of the terminal: 
[phoebus.log](https://github.com/ControlSystemStudio/phoebus/files/4381062/phoebus.log)

georgweiss, 2020-03-25:
The log suggests two different problems:
1) You are trying to save a snapshot under the same name as an existing one. This is not allowed.
2) You are trying to restore a snapshot, but the PV cannot be write connected. Is the PV writable or read-only?

dsoklic, 2020-03-26:
I don't think this is connected with the disconnected PV issue. This issue already exists when I only create a saveset and open it up. The Live Setpoint is already disconnected at this point.

Here is a log where nothing else is pressed: 
[phoebus2.log](https://github.com/ControlSystemStudio/phoebus/files/4385722/phoebus2.log)

georgweiss, 2020-03-26:
You see this behaviour irregularly, correct? When this happens, are you able to connect to the PV through an epics tool like caget?

geniejhang, 2020-03-26:
I also saw VEnum type PVs are disconnected all the time while I'm testing.
It looks like the PV with issue is also VEnum type.

Try this one with softIoc.

record(bi, "STATUS") {       
  field(DTYP, "Soft Channel")
  field(ZNAM, "OFF")
  field(ONAM, "ON")
  field(PINI, 1)
}

georgweiss, 2020-03-26:
Thanks @geniejhang for the tip. 
Trying with VEnum the epics lib (GPClient) gets an exception.
@shroffk , is VEnum not supported in GPClient? In the save-and-restore client we're getting:
{Type: [EXCEPTION] - ex: DataSource misconfiguration: no match found to convert payload to type. (class org.epics.vtype.VType - CAConnection [connected: true writeConnected: true channel: CAJChannel = { name = STATUS, connectionState = CONNECTED }])}

The CONNECTED state is misleading as the event says "disconnected".

dsoklic, 2020-03-27:
@geniejhang thank you for this discovery.

It seems like my issue has two sides to it. Unbeknownst to me, I was using a fork that was based on version 4.6.0. After discovering this and updating to the latest version on the master branch, my original issue is fixed. Sorry about that, @georgweiss

I can however still reproduce this issue when using VEnum .

georgweiss, 2020-03-27:
Thanks for the update.
I will dig into the VEnum issue next, but please expect delays.

georgweiss, 2020-04-04:
Update: I can confirm that the gpclient library used to interact with the IOC from the save-and-restore app indeed lacks support for VEnum. I will update that library, but this may take some time as other changes are scheduled as well for a new version of gpclient.

geniejhang, 2020-05-20:
This one is resolved I think.

closed 2020-05-21 by georgweiss