ControlSystemStudio/phoebus #1238: Display Builder Runtime: DisplayRuntime startup stucks in PVPool

pavel-ch, 2020-03-28:
Irregularly, the DisplayRuntime.start() method (usually on Embedded Widgets) gets stuck in PVPool.getPV. See also the generated exception in console.
This happens to me sometimes (not seen any regular behaviour) when using more formula PVs like `='pv1' + 2 * 'pv2'`
It seems to me that in the case of formulaPV with PV nodes, we are trying to modify the `ConcurrentHashMap` inside `PVPool` from inside its `computeIfAbsent` method (called once for formula PV and then again from same thread for PV nodes from inside this function). For the `computeIfAbsent` function documentation says is that:
     * If the value for the specified key is present, attempts to
     * compute a new mapping given the key and its current mapped
     * value.  The entire method invocation is performed atomically.
     * Some attempted update operations on this map by other threads
     * may be blocked while computation is in progress, so the
     * computation should be short and simple, and **must not attempt to
     * update any other mappings of this map**.

Can this be a reason for observed behavior?
For sure in my code there is a lot of other chances being buggy and cause this, but the debugger pointed me to the PVPool - see Eclipse printscreen pictures showing end of stack. Last picture shows the point of lock: ConcurrentHashMap:2477 synchronized(f)

Could you please look at that Kay @kasemir ? I am not experienced enough to see the problem.
In case this really is a problem with FormulaPV this can save potential pain to all.
[image]
[image]
[image]
[image]
[image]

kasemir, 2020-03-30:
I think you're right. A formula like
```
=`A` + `B`
```
will result in recursively adding `A` and then `B` to the PVPool while it's adding the overall formula itself, and `computeIfAbsent` doesn't allow that. I'll need some time to look for a test that would demonstrate the lockup, since it's clearly not happening every time.

kasemir, 2020-03-30:
Do you see `java.lang.IllegalStateException: Recursive update .. at .. ConcurrentHashMap.computeIfAbsent` in the console output? Or is it always a lockup, no error message from the map?

pavel-ch, 2020-03-30:
In my setup, only the lockup was seen. Never this RecursiveUpdate ... output. I am also totally confused with the irregularity of the occurrence.

kasemir, 2020-03-30:
Well, the way it called `ConcurrentHashMap.computeIfAbsent` was clearly problematic, I have a test that fails every time with that IllegalStateException, and just committed a fix.
We'll have to see if that also fixes the slightly different failure mode that you see.

pavel-ch, 2020-03-30:
Thanks I will give it a try. In these edge cases, maybe the difference between Windows and Linux JVM implementation may play a role?

kasemir, 2020-03-30:
The 'Recursive update' exception was caught/printed, maybe it did show up somewhere earlier in your console output?

As for the hangup, that's hard to duplicate in a test since it requires two threads to create formula PVs are exactly the wrong time, which could happen with embedded displays that are all loaded in parallel. The test I used showed a problem with ConcurrentHashMap.computeIfAbsent being called recursively on a single thread, so I think it's demonstrating the same root cause, which was then fixed.

So give it a try, and then we'll see if there's yet another variant of the issue.

pavel-ch, 2020-03-31:
Seems that the effect has gone. Due to its random nature, I do not have a test which would always fail or succeed. But I tried most situations when it failed, an now I have no one occurrence of the lockup. I would say the root cause has been solved.
Thank you!

As for 
> The 'Recursive update' exception was caught/printed, maybe it did show up somewhere earlier in your console output?

no this has not been observed

kasemir, 2020-03-31:
Thanks for testing it; good news.
I think we can close this; re-open in case we get new faults.

closed 2020-03-31 by kasemir