ControlSystemStudio/phoebus #1242: Save And Restore: Stored Setpoint cell editing

martinapav, 2020-03-31:
We want to change »Stored Setpoint« in Save And Restore table. By clicking on the Stored Setpoint cell we can edit it. After changing stored set point to 5 and pressing enter also »Δ Live Setpoint« changes.
 
[image]

If we click again into the cell and only press enter without entering any different value the value goes back to previous setting but »Δ Live Setpoint« does not change.
 
[image]

If I only edit the value and click Save snapshot, Live Setpoint seems to be saved. Is this field intended to be editable at all?

georgweiss, 2020-03-31:
No, not initially. I have other requests for making the stored setpoint editable (and possible to persist), but that has not been implemented.
Your observation points to an inconsistency. What I can do short term is to make the cell read-only.

geniejhang, 2020-06-12:
I'm working on this for our application. Just FYI.

georgweiss, 2020-06-12:
Thanks @geniejhang. 
There is one twist to this that I'd like us to agree upon: if the cell value is changed before restore, it must be clear to the user that the changed value is not saved in the snapshot by default. Doing so would in my view be very confusing and in part defeat the purpose of a snapshot. To persist the changed value in the snapshot should require the user to press the save button. Also, if the snapshot tab is closed without saving the value, the user should be prompted to choose to exit without save.

geniejhang, 2020-06-12:
@georgweiss 
Yes. I was thinking about that last night lying on the bed.
FRIB version of CS-Studio SaveSetRestore has pink highlight when the saveset/snapshot becomes modified state. I'm thinking about putting similar stuff in as well.
I agree on the behavior with you.
Thanks for the comments.

geniejhang, 2020-06-26:
@martinapav Since it's now merged to master, please test it 😄