ControlSystemStudio/phoebus #1246: Custom cursor images fail to render on Mac

kasemir, 2020-03-31:
Some tools set custom JavaFX `ImageCursor`, like the plot toolbar options for zoom/pan which set the cursor to match the option.

On Mac OS X, that might work when the cursor is first selected, but after some time the cursor turns into a random image, like shown here below the toolbar option that it should represent:

[image]

Have only observed this on Mac OS.

georgweiss, 2020-04-01:
I'm afraid the same thing sometimes happens to menu item icons. On Mac.

kasemir, 2020-04-01:
With menu items, I see a scaling issue.

[image]

The "Data Browser" icon is available as both `databrowser.png` and `databrowser@2x.png`. In most cases, the JFX `Image` class automagically picks the higher resolution icon to show more detail on hires monitors without changing the on-screen image size (https://dlsc.com/2017/08/29/javafx-tip-27-hires-retina-icons/, #65).
For use as menu icons, however, that backfires and the hires image is shown too large.

Is that what you see, or do you get erroneous noise as in the cursor screenshot for a menu icon?

georgweiss, 2020-04-01:
I'm seeing both. I have ignored the scaling issue as I have seen it only on my 13" laptop, which is not a prioritized use case. On the other hand it is reproducible. Same issue in Utility menu.
The noisy icon issue happens quite rarely in menus.
[image]

kasemir, 2020-04-01:
For the cursor, some initial tests suggest that I get a correct custom cursor if I keep setting a `new ImageCursor(same_image, hotspot_x, hotspot_y)`, i.e. reuse the image, but keep creating a new `ImageCursor` instead of just creating that cursor once and re-using it.

But if you see erroneous noise for menu icons, that suggests that the Mac implementation also forgets the content of images? Try to get a screenshot of the menu icon issue.

georgweiss, 2020-04-01:
I will, but this make take time. However, the appearance is the same as for the custom cursor.

closed 2020-04-06 by kasemir