ControlSystemStudio/phoebus #1254: Save & Restore: livesetpoint NaN PV storing error

geniejhang, 2020-04-07:
I tried to save NaN livesetpoint PV to DB and it spits an error.

Possibly it's coming when converting PV data to JSON?

Here's the one coming from the application
[image]

and service
2020-04-07 11:52:28.940 [http-nio-8080-exec-5] WARN  org.springframework.web.servlet.mvc.support.DefaultHandlerExceptionResolver - Resolved [org.springframework.http.converter.HttpMessageNotReadableException: JSON parse error: (was java.lang.ClassCastException); nested exception is com.fasterxml.jackson.databind.JsonMappingException: (was java.lang.ClassCastException) (through reference chain: java.util.ArrayList[0]->org.phoebus.applications.saveandrestore.model.SnapshotItem["value"])]

Thanks.

georgweiss, 2020-04-07:
Well, question is what to expect. Saving a NaN setpoint does not make sense as it cannot be restored.

kasemir, 2020-04-07:
> it cannot be restored.

It can. `caput SomePVName NaN` writes not-a-number.
The EPICS `epicsStrnCaseCmp` routine specifically recognizes 'NaN', 'Inf'.

geniejhang, 2020-04-07:
YES! I though it's not the real case scenario as well. But, our operator says it is.

kasemir, 2020-04-07:
I agree that it's a corner case, a lot of device support might not handle Double.NAN or inf., but EPICS fundamentally allows numeric records to have those values.

georgweiss, 2020-04-07:
But my point is: does it make sense to restore a NaN value?

kasemir, 2020-04-07:
In the CS-Studio/Phoebus 'Probe' and 'PV Table' apps you can also enter 'NaN' as values and write those to e.g. an analog output record. Note that in there you must write 'NaN'. Just 'NAN' or 'nan' doesn't work in the PV table.

Does it make sense to restore NaN?
If that was saved, then it was obviously the original value, and I'd suggest it needs to be restored.
For example, some device support for a power supply might turn the PS off when the setpoint is NaN.

georgweiss, 2020-04-07:
I see, will fix this.

georgweiss, 2020-04-07:
For the sake of testing, how can I create a soft IOC or datasource that sets a NaN value?

kasemir, 2020-04-07:
```
# File demo.db
#
# Run with `softIoc -d demo.db`,
# then `caput test NaN` to set to, well, NaN
record(ao, "test")
{
  field(PINI, "YES")
}
```

georgweiss, 2020-04-08:
So this is an interesting case...
Turns out the culprit is in the vtype-json lib that serializes a NaN value to null. The fix is to serialize to "NaN" and check for this string when converting back to a VType. One could also consider supporting infinity, but since there is Double.NEGATIVE_INFINITY and Double_POSITIVE_INFINITY we are faced with the problem of choosing...
In any case, a new release of vtype-json assumes a new release of epicsCoreJava, which we need anyway to support VEnum in GPClient. 
I'd like  @kasemir and @shroffk to comment.

closed 2020-05-14 by georgweiss