ControlSystemStudio/phoebus #1262: [Question] create new widget

CC-g2ca, 2020-04-13:
Hello,
I would like to have more information in order to add some widgets (specifically for our laboratory). Before, on CSS there were three components (model, figures and editparts). We would now like to know what these components correspond to. And if you have more information or docs to help us to create widgets, we are interested about that.

kasemir, 2020-04-13:
See https://github.com/ControlSystemStudio/phoebus/blob/master/app/display/Readme.md

Same idea, but instead of Eclipse extension points we use plain SPI.
The basic model class is "Widget". Compared to before we have a clear separation of model, representation and runtime, whereas figure and edit parts mixed the latter two up.
You know need a "Widget" to describe the properties, a "Representation" like the "EllipseRepresentation" to render the model on the screen. Some widgets also need a "Runtime" to react to PV changes, but most can use the default runtime which connects the "pv_name" to the "pv_value".

CC-g2ca, 2020-04-13:
thanks for your information.

closed 2022-03-03 by kasemir