ControlSystemStudio/phoebus #1269: LED font color doesn't display correctly

ihtao127, 2020-04-16:
I realy like the feature that the LED font changes the color base on the backgroud color, if the LED color is dark, the font changes to a white color, it is good for the readability.

But if the state labels of the LED are the same, this feature doesn't work anymore.

Please look at _**[the video](https://youtu.be/dzVs8tpLde4)**_.

kasemir, 2020-04-16:
Indeed, you're correct, it's about the label. When the label doesn't change, the color isn't recomputed. Thanks for figuring that out!
Will fix it soon.

kasemir, 2020-04-16:
For what it's worth, the labels are meant to help users with color vision issues. So it's somewhat important to have different labels to clearly indicate the state.
Still, the bug when using the same label has been fixed.

closed 2020-04-20 by kasemir