ControlSystemStudio/phoebus #1273: Inconsistent text alignment in CS-Studio phoebus Widgets

epicDirk, 2020-04-20:
[bob_files.zip](https://github.com/ControlSystemStudio/phoebus/files/4502157/bob_files.zip)
[image]
[image]

The text alignment is inconsistent in CS-Studio phoebus Widgets. This lowers the readability and situation awareness.

The ideal vertical text alignment would be if a capital letter had the same distance on the top and bottom to the widgets border in all widgets.

kasemir, 2020-04-20:
> same distance on the top and bottom

For that, you can select "Vertical Alignment: Middle". With the default top-alignment, the exact location of the baseline depends on the font, and then comparisons are difficult when you don't have the exact same fonts.
Like the "Text Update, Entry and Spinner" label in your example asks for the font "Source Sans Pro", which I don't have, so for me it looks like this:

[image]

epicDirk, 2020-04-21:
[image]
"Vertical Alignment: Middle" does not work if the text update is just 20 pixel high. Even with a font like Arial at size 16 for example. My workaround is to make the text update 40 pixel high because then the "Vertical Alignment: Middle" works. But the disadvantage is then that the text updates overlaps. Maybe a solution would to make the internal size 40 when Vertical Alignment is set to Middle even if it is just set to 20 in high. But i am not a software developer just a user.

kasemir, 2020-05-05:
> Which font do you use?

We tend to stick to the default "Liberation Sans" font, since it's bundled with the Display Builder, so you get very similar results on all platforms. "Arial" might be common on Windows computers, but not Linux.

epicDirk, 2020-06-03:
With the "Liberation Sans" the Inconsistent text alignment is less visible but still there.
[image]