ControlSystemStudio/phoebus #1305: Alarm table/tree: open display with macros error

andrej-m, 2020-05-07:
Opening a display from the PV drop-down menu in alarm table/alarm tree doesn't work if macros are specified:
```
2020-05-07 11:06:45 WARNING [org.phoebus.framework.jobs] Job 'Open display /w macros' failed
java.io.IOException: Cannot run program "/home/andrej/test/test.bob" (in directory "/home/andrej"): error=13, Permission denied
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1128)
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1071)
        at org.phoebus.framework.jobs.CommandExecutor.call(CommandExecutor.java:112)
        at org.phoebus.applications.alarm.ui.OpenDisplayAction.lambda$new$1(OpenDisplayAction.java:74)
        at org.phoebus.framework.jobs.Job.execute(Job.java:50)
        at org.phoebus.framework.jobs.JobManager.execute(JobManager.java:54)
        at org.phoebus.framework.jobs.JobManager.lambda$schedule$0(JobManager.java:45)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.io.IOException: error=13, Permission denied
        at java.base/java.lang.ProcessImpl.forkAndExec(Native Method)
        at java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:340)
        at java.base/java.lang.ProcessImpl.start(ProcessImpl.java:271)
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1107)
        ... 10 more
```
[image]

Here's the alarm PV configuration:
[image]

Is my syntax for specifying macros wrong?

Opening the same display file without macros works nicely.

kasemir, 2020-05-07:
> Is my syntax for specifying macros wrong?

Certainly.

See generic format of URLs in launcher '-help' output, https://control-system-studio.readthedocs.io/en/latest/running.html#command-line-parameters-for-applications,
or check the "Info" context menu of a display file's tab for the input format.

It's something like this:
```
/path/to/file?PREFIX=test&EXAMPLE_MACRO=Value+with+spaces&X=u&N=42
```

kasemir, 2020-05-07:
I'll look at adding more detail to the alarm info, https://control-system-studio.readthedocs.io/en/latest/app/alarm/ui/doc/ is indeed somewhat sparse at this time.

kasemir, 2020-05-07:
Added display-with-macro example to the alarm doc, #1308 
```
/path/to/display.bob
/path/to/display.bob?MACRO=Value&OTHER=42$NAME=Text+with+spaces
```

andrej-m, 2020-05-13:
Thanks for looking into it Kay.

I updated the URL, still no success, however the error message is now different:
```
java.io.IOException: Cannot run program "/home/andrej/test/test.bob?PREFIX=test" (in directory "/home/andrej"): error=2, No such file or directory
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1128)
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1071)
        at org.phoebus.framework.jobs.CommandExecutor.call(CommandExecutor.java:112)
        at org.phoebus.applications.alarm.ui.OpenDisplayAction.lambda$new$1(OpenDisplayAction.java:74)
        at org.phoebus.framework.jobs.Job.execute(Job.java:50)
        at org.phoebus.framework.jobs.JobManager.execute(JobManager.java:54)
        at org.phoebus.framework.jobs.JobManager.lambda$schedule$0(JobManager.java:45)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.io.IOException: error=2, No such file or directory
        at java.base/java.lang.ProcessImpl.forkAndExec(Native Method)
        at java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:340)
        at java.base/java.lang.ProcessImpl.start(ProcessImpl.java:271)
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1107)
        ... 10 more

```

Additionally, trying to open the display from CLI with `-resource` argument also doesn't work if I specify macros:
```
$ ./phoebus.sh -resource "/home/andrej/test/test.bob?PREFIX=test"
2020-05-13 14:16:04 INFO [org.phoebus.product.Launcher] Phoebus (PID 17224)
2020-05-13 14:16:06 INFO [org.phoebus.channelfinder.ChannelFinderClient] Creating a channelfinder client to : http://localhost:8080/ChannelFinder
2020-05-13 14:16:06 INFO [org.phoebus.applications.display.navigation.DisplayNavigationViewApp] Loading Display Navigation View
2020-05-13 14:16:08 WARNING [org.phoebus.ui.application.PhoebusApplication] No application found for opening file:///home/andrej/test/test.bob%3FPREFIX=test
```

kasemir, 2020-05-13:
Regarding the command line `-resource...`, note how the '?' gets URL-escaped into `%3F` when the `/home/...` path gets turned into a pukka URL `file://...`?
You can avoid that by passing a proper URL from the start. Depending on your shell, you need to further escape resp. quote it. Try this:
```
$ ./phoebus.sh -resource 'file:///home/andrej/test/test.bob?PREFIX=test'
```
Note single-quotes to prevent shell from interpreting it, and file-colon-triple-slash.

Need to look into the use from UI.

kasemir, 2020-05-13:
In the alarm UI, when using macros, a complete URL like "file:///home/andrej/test/test.bob?PREFIX=test&X=u&ANSWER=42" should work. In there, compared to the command line, no need to add quotes since the text is used as entered, no shell to interpret it.

kasemir, 2020-05-13:
Updated online doc to give alarm display examples, mention that macros require using URL.

andrej-m, 2020-05-21:
Specifying the URL in `file://...` format works in both cases. Thanks for your help!

[image]

[image]

closed 2020-05-21 by andrej-m

MaticPogacnik, 2021-03-04:
@kasemir 
I may have a similar problem, I want to construct the `-resource` parameter dynamically.
But I can not find a way to do that as the bash or Phoebus keep expending the path.

My script, what I want to do (variables comes from the Enviromint or user):

```
res="${DASHBOARD_PATH}?app=display_runtime&CONFIG=${CONFIG_PATH}"
echo $res

daq-css-phoebus -layout "" -resource $res
```

Echo output is as following:

```
../pkgs/gateway_dashbaord-0.0.2-py38_0/lib/python3.8/site-packages/bob/gateway_dashboard.bob?app=display_runtime&CONFIG=/opt/anl_develop/gateway_gui/bob/gateways.json
```

But the bash or Phoebus later expend the `?` somehow to `%3` (otherwise the path is correct):

```
2021-03-04 16:40:07 WARNING [org.phoebus.ui.application.PhoebusApplication] No application found for opening file:///home/devman/miniconda3/envs/gateway_dashboard/bin/../pkgs/gateway_dashbaord-0.0.2-py38_0/lib/python3.8/site-packages/bob/gateway_dashboard.bob%3Fapp=display_runtime&CONFIG=/opt/anl_develop/gateway_gui/bob/gateways.json
```

Is there any other solution to this beside keeping everything "fixed" and in single quotes?
I would apreciate any tips, thanks.