ControlSystemStudio/phoebus #1309: Selection behavior in display editor

georgweiss, 2020-05-08:
When selecting multiple objects in the display editor a selection area is drawn. The limits of this bounding box is defined by the lowest and highest coordinates found among the selected items. There is a side effect of this: additional objects within that rectangle cannot be added to the selection using keyboard and mouse in the editor area, user will need to add such selections in the widgets table on the left.

I'd like to understand the rationale of this behavior as it contrast to other applications (e.g. PowerPoint) where the selection is rendered by highlighting only the selected objects, and not the bounding box defined by (some of) them.

pavel-ch, 2020-05-11:
Adding / Removing object inside selection area is really something which should work, regardless of the implementation (one or more selection box(es)). Right now, removing works but adding not.

kasemir, 2020-05-11:
Yes, it's based on the 'EDM' idea where the overall selection size is clearly visible, there are easy to grab resize handles (no need to try and hit the exact line to resize). To change the selection of widgets, i.e. to add or remove a widget that's under the tracker outline, you need to use control-click instead of a plain click.
.. Except for some reason that's broken for addition. Control-click will remove a widget from the selection, but not add it (Command-click on Mac since Control-click is already used for context menu on Mac). There's a bug somewhere in the event filter/handler.

kasemir, 2020-05-11:
Fixed.
The tracker already had 'passClickToWidgets' because of an issue with passing events on to underlying widget. A JavaFX event _filter_ (as opposed to _handler_) simply didn't work, events with Ctrl or Command pressed didn't reach the widgets, so the Tracker did that on its own.
.. but only for widgets in the selection, so those could be de-selected.
Updated to check all widgets, so now you can Ctrl (Command)-click to add as well as remove from selection.

pavel-ch, 2020-05-11:
Thank you Kay

kasemir, 2020-05-11:
It's not recursive, yet, so doesn't fully work within for example a 'group'. Will have another update later today.

georgweiss, 2020-05-11:
Well, not sure what to expect, but I still cannot add a widget inside the "tracker" box. Attached BOB file is an example: if you select (using CMD + click on Mac) buttons 1,2 and 4, then button 3 cannot be added to the selection using CMD + click.

[image]
[select.bob.zip](https://github.com/ControlSystemStudio/phoebus/files/4611194/select.bob.zip)

kasemir, 2020-05-11:
Select 1, 2, 3:

[image]

Command-click on 4 to add it to the selection:
[image]

Cmd-click again on 4, and it's removed from the selection as in the first screenshot.

georgweiss, 2020-05-12:
I can confirm, thanks @kasemir

closed 2020-05-12 by georgweiss