ControlSystemStudio/phoebus #1317: Slider decrement bug

kingspride, 2020-05-12:
Hi devs,

thought I'd create a bug report on here instead of writing an email to kay.

okay, so today we discovered a bug  with the scaled slider widget.

If you want to decrement via click on the bar, left to the knob, 
it will only decrement when you click far enough to the left - otherwise it will increment instead of decrement. this seems to "walk" with the current value / knob position.
its especially hard to successfully decrement when the PV is negative.

I'll attach a demo video, bc its hard to explain but easy to watch ;)

https://cloud.fhi-berlin.mpg.de:8443/getlink/fi21jr781qcbhMozHCxnHtxa/2020-05-12_15-50-19.mkv

kasemir, 2020-05-14:
Cannot reproduce on Mac OS nor Linux.
Do you see that with controls_spinner_slider_scrollbar.bob from the examples?

Details can depend on the PV.
Moving the slider writes to the PV.
But something else could also write to the PV, and in that case the slider needs to be moved, i.e. data flows the other way.
So moving the slider writes to the PV, and ideally we get that written value back from the PV ASAP and everybody is happy. Otherwise, after a short timeout, the slider it moved back to the current value of the PV, since the PV didn't like what we wrote and we need to reflect the actual value.

kingspride, 2020-05-15:
It appearently only happens when using negative numbers?

I used a minimum of -4 and a maximum of 4 and a local PV.

bob is attached.
[broken-slider.zip](https://github.com/ControlSystemStudio/phoebus/files/4635130/broken-slider.zip)

kasemir, 2020-05-19:
Good example! I can reproduce the error with that one, will look into it.

kasemir, 2020-05-19:
Found it, again having a simple example made it easy.

closed 2020-05-26 by kasemir