ControlSystemStudio/phoebus #1323: Nice to have in Data browser: min/max/avg/dev info to each trace

mvimercati, 2020-05-13:
Hello,
it would be nice to have for each plot trace some important info such as 
* min value
* max value
* average
* standard deviation 
at least for archived data.
Do you think it is feasable?

kasemir, 2020-05-14:
If your archive data source provides that information, it's shown.

For the RDB archive data source, you get min/max/average.
As described in the online help, the colored outline reflects the min/max, and the line represents the average:

[image]

If the archive data source also has std deviation, that's shown via another thinner line:

[image]

mvimercati, 2020-05-18:
I read about this feature but I understood that it was related to downsampling/binning the retrieved data from archive. What I mean is a simple average all over the selected period, so it would be one number information for all the chart.
Am I missing something?
BTW thanks for you reply

kasemir, 2022-03-03:
The data browser properties panel has a "Statistics" tab, https://github.com/ControlSystemStudio/phoebus/pull/1532 :

[image]

closed 2022-03-03 by kasemir