ControlSystemStudio/phoebus #1326: Macro editor dialog triggers error dialog

georgweiss, 2020-05-14:
The macro editor dialog triggers the attached error dialog in the following case:
1) User enters a value for the macro name and hits ENTER -> value input field gets focus and can be edited.
2) User enter a value for the macro value and hits ENTER -> new row is created and the input field for macro name gets focus.
3) User does not wish to add additional macros and hits OK button -> Error dialog triggered as default macro name is invalid.

[image]

shroffk, 2020-05-14:
I think we could remove the automatic move to the next row.

User presses ENTER.. the value edit is submitted and the row is selected.
TAB will result in the "Add" button being selected.
ENTER will result in start the editing of a new MACRO.

tanviash, 2020-05-14:
@shroffk I don't anticipate any issue with the new workflow so okay for me.

shroffk, 2020-05-14:
I have added a patch but we will wait till everyone agrees before I make a PR

kasemir, 2020-05-14:
The actual problem is the "<Enter ...>" hint, not the navigation.
The navigation, i.e. the automated move to the next field on enter, is very convenient when you need to add several macros and values. Similar functionality also makes the rule dialog "flow" since you get from the rule name to the PVs to the expressions.

I'd vote for just removing the "<Enter ...>" hint, and keeping the navigation.

shroffk, 2020-05-14:
I see the benefit in having both the flow and the hint.

I was testing this above issue on the old master. After the #1324 was resolved, the above issue was also resolved as part of the same fix.

So we can keep the hint (which is great for guiding users)
&
Allow the flow 

@georgweiss can you confirm that the lastest master does not have the above issue.

georgweiss, 2020-05-14:
This fixes the #1326, but there is another (new?) corner case: if user selects macro name and hits ENTER without typing a value, an exception is thrown:
`Exception in thread "JavaFX Application Thread" java.lang.IndexOutOfBoundsException: Index 0 out of bounds for length 0
	at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:64)
	at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:70)
	at java.base/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:248)
	at java.base/java.util.Objects.checkIndex(Objects.java:372)
	at java.base/java.util.ArrayList.get(ArrayList.java:458)
	at com.sun.javafx.collections.ObservableListWrapper.get(ObservableListWrapper.java:89)
	at org.csstudio.display.builder.representation.javafx.MacrosTable.lambda$new$6(MacrosTable.java:152)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:234)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:49)
	at javafx.event.Event.fireEvent(Event.java:198)
	at org.phoebus.ui.javafx.EditCell.commitEdit(EditCell.java:114)
	at org.phoebus.ui.javafx.EditCell.lambda$new$2(EditCell.java:46)
	at com.sun.javafx.binding.ExpressionHelper$Generic.fireValueChangedEvent(ExpressionHelper.java:360)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:80)
	at javafx.beans.property.ReadOnlyBooleanPropertyBase.fireValueChangedEvent(ReadOnlyBooleanPropertyBase.java:72)
	at javafx.scene.Node$FocusedProperty.notifyListeners(Node.java:8154)
	at javafx.scene.Scene$12.invalidated(Scene.java:2194)
	at javafx.beans.property.ObjectPropertyBase.markInvalid(ObjectPropertyBase.java:112)
	at javafx.beans.property.ObjectPropertyBase.set(ObjectPropertyBase.java:147)
	at javafx.scene.Scene$KeyHandler.setFocusOwner(Scene.java:4058)
	at javafx.scene.Scene$KeyHandler.requestFocus(Scene.java:4105)
	at javafx.scene.Scene.requestFocus(Scene.java:2161)
	at javafx.scene.Node.requestFocus(Node.java:8315)
	at javafx.scene.control.ControlUtils.requestFocusOnControlOnlyIfCurrentFocusOwnerIsChild(ControlUtils.java:64)
	at javafx.scene.control.TableCell.cancelEdit(TableCell.java:392)
	at org.phoebus.ui.javafx.EditCell.cancelEdit(EditCell.java:96)
	at javafx.scene.control.TableCell.updateEditing(TableCell.java:567)
	at javafx.scene.control.TableCell.lambda$new$3(TableCell.java:141)
	at javafx.beans.WeakInvalidationListener.invalidated(WeakInvalidationListener.java:83)
	at com.sun.javafx.binding.ExpressionHelper$Generic.fireValueChangedEvent(ExpressionHelper.java:348)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:80)
	at javafx.beans.property.ReadOnlyObjectPropertyBase.fireValueChangedEvent(ReadOnlyObjectPropertyBase.java:74)
	at javafx.beans.property.ReadOnlyObjectWrapper.fireValueChangedEvent(ReadOnlyObjectWrapper.java:102)
	at javafx.beans.property.ObjectPropertyBase.markInvalid(ObjectPropertyBase.java:113)
	at javafx.beans.property.ObjectPropertyBase.set(ObjectPropertyBase.java:147)
	at javafx.scene.control.TableView.setEditingCell(TableView.java:1175)
	at javafx.scene.control.TableView.edit(TableView.java:1494)
	at javafx.scene.control.skin.TableViewSkinBase.lambda$new$2(TableViewSkinBase.java:206)
	at javafx.collections.WeakListChangeListener.onChanged(WeakListChangeListener.java:88)
	at com.sun.javafx.collections.ListListenerHelper$Generic.fireValueChangedEvent(ListListenerHelper.java:329)
	at com.sun.javafx.collections.ListListenerHelper.fireValueChangedEvent(ListListenerHelper.java:73)
	at javafx.collections.ObservableListBase.fireChange(ObservableListBase.java:233)
	at javafx.collections.ListChangeBuilder.commit(ListChangeBuilder.java:482)
	at javafx.collections.ListChangeBuilder.endChange(ListChangeBuilder.java:541)
	at javafx.collections.ObservableListBase.endChange(ObservableListBase.java:205)
	at javafx.collections.ModifiableObservableListBase.remove(ModifiableObservableListBase.java:183)
	at org.csstudio.display.builder.representation.javafx.MacrosTable.fixup(MacrosTable.java:294)
	at org.csstudio.display.builder.representation.javafx.MacrosTable.lambda$new$6(MacrosTable.java:158)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:234)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:49)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.control.TableCell.commitEdit(TableCell.java:350)
	at org.phoebus.ui.javafx.EditCell.commitEdit(EditCell.java:118)
	at org.phoebus.ui.javafx.EditCell.lambda$new$1(EditCell.java:41)
[... cut ...]
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$2(GlassViewEventHandler.java:447)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:412)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:446)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:556)
	at com.sun.glass.ui.View.notifyMouse(View.java:942)
	at com.sun.glass.ui.mac.MacView.notifyMouse(MacView.java:127)
`

shroffk, 2020-05-14:
Thanks for confirming this fix.
I have deleted the above changes to the flow and restored the old way in which the dialog worked.

About the new error, I reverted back to before I made any changes and I was able to trigger this same error. So this is a different error, we can keep this Issue open to address it here.

closed 2020-05-19 by georgweiss