ControlSystemStudio/phoebus #1338: Alarm Table: show hint when hovering over buttons

mark0n, 2020-05-20:
It might not be clear to everyone what these buttons are for:
[image]
Please provide a hint.

tanviash, 2020-05-21:
@mark0n The maintenance mode & disable_notification mode buttons show a tooltip when they are enabled based on permissions.

mark0n, 2020-05-21:
Hmm, seems like JavaFX has an interesting philosophy here. Anyway, the more I'm thinking about it the more I realize that the usability issue might actually be more related to the icons and the logic behind the buttons:

1. Some of the icons aren't exactly self-explanatory. At least on my machine the mail icon looks more like a yellow rectangle (see screen-shot above). I'm wondering if this is a consequence of it being scaled down. Others like "Collapse alarm tree" and "Expand alarm tree to show active alarms" aren't simple enough to be easily recognized. Have you considered using simpler icons like triangles pointing down/up and for cases where such simple icons do not exist or aren't clear enough using text instead (or at least icon + text)?
2. I'm confused by the logic of the buttons: The email notification button shows an envelope with a little "X". I'm expecting that clicking this button would disable the notifications. However, what actually happens is the opposite! Shouldn't the icon/text on a button describe what _will happen_ when I click it rather than the _current status_ of the program?
3. Have you considered showing the current status somewhere else? Maybe in a status bar?
4. I would recommend using positive language. "[X] Enable email notification" rather than "[ ] Disable email notification".
5. Toggle buttons can also simply use highlighting of the background to show that a feature is active:
[image]
Microsoft provides some good [advise on when to use toggle switches](https://docs.microsoft.com/en-us/windows/uwp/design/controls-and-patterns/toggles#choosing-between-toggle-switch-and-check-box).

While we're at it: The current UI has more quirks. For example

- The status bar is showing the number of acknowledged alarms above the list of acknowledged alarms rather than below (see screen-shot above). This is not where I would expect it.
- The number of active alarms is displayed above the list of pending alarms. Status and controls are mixed here. To make matters worse the status changes when I select a different alarm server. This breaks left to right order...

shroffk, 2022-04-08:
@tanviash do we have any update on this issue? are we planning on making any changes

tanviash, 2022-04-08:
I'm closing this issue ticket.

closed 2022-04-08 by tanviash