ControlSystemStudio/phoebus #1339: Save and restore: restoring arrays

willrogers, 2020-05-21:
I got a demo save and restore service working, but I have a problem with arrays:

If I save an array of length three, it correctly shows [1.0, 2.0, 3.0] in the UI. However, if I change the value of the PV then press the Restore button the array value doesn't change back. It seems to work ok for scalar values.

Any ideas what's going on here @shroffk @georgweiss ?

georgweiss, 2020-05-21:
Will investigate.

willrogers, 2020-05-22:
This is what the UI looks like.

[image]

willrogers, 2020-05-22:
This is what the logging looks like when I press Restore, with an additional print that I added:

```
2020-05-22 11:52:10 INFO [org.phoebus.applications.saveandrestore.ui.snapshot.SnapshotController] java.util.concurrent.CountDownLatch@2ec115ad[Count = 1] New CountDownLatch set
Val: VDoubleArray[[1.0, 2.0], size [2], NONE(NONE) - NONE, 2020-05-22T10:50:08.622124324Z]
2020-05-22 11:52:10 INFO [org.phoebus.applications.saveandrestore.ui.snapshot.SnapshotController] java.util.concurrent.CountDownLatch@2ec115ad[Count = 1] Write OK, signalling latch
2020-05-22 11:52:10 FINE [org.phoebus.applications.saveandrestore.ui.snapshot.SnapshotController] Restored snapshot aaa
```

georgweiss, 2020-05-26:
For the sake of clarity: is the size of the array unchanged between save and restore? The size is 2?

georgweiss, 2020-05-28:
To fix this we need to update the gpclient dependency, which may take a week or two.
Do you need a workaround in the meanwhile?

kasemir, 2020-05-28:
Note that save/restore is the only phoebus app that uses the gpclient.
The rest uses phoebus.pv, which reads & writes arrays.
Instead of adding & debugging a different PV library, you could switch the the common one.

georgweiss, 2020-05-28:
I have already considered that, though for the moment I do not have time to do a switch.

geniejhang, 2020-06-02:
@mark0n Is this the same issue that you told me before?

georgweiss, 2020-06-04:
Fixed through PR #1371.

closed 2020-06-04 by georgweiss

willrogers, 2020-06-04:
Thanks a lot, this seems to be working fine now.

mark0n, 2020-06-04:
@geniejhang the issue I saw was related to saving. It silently failed to retrieve an array when it was larger than `EPICS_CA_MAX_ARRAY_BYTES`. The default in EPICS is 16 kB, CS-Studio might even default to more. The array mentioned here is only a few bytes.

georgweiss, 2020-06-05:
@mark0n , I will fix this.

georgweiss, 2020-06-05:
@mark0n , @geniejhang , I have merged a fix for EPICS_CA_MAX_ARRAY_BYTES. I have verified it for small arrays, I would appreciate if you can verify with data > 16 kB.