ControlSystemStudio/phoebus #1344: How does the alarm delay/count work?

MaticPogacnik, 2020-05-22:
Hi,

I am testing alarm delay and count options, but they doesn't seem to work for me.

[image]

For example, I have Delay of 7 seconds on this PV, but it still enters alarm state, even if it exits alarm state in this time.

shroffk, 2020-05-22:
@MaticPogacnik can you give some details on what we are seeing in the 3rd screen?
Is that camonitor?

MaticPogacnik, 2020-05-22:
@shroffk Sorry, yes 3rd screens is camonitor for that PV.

shroffk, 2020-05-22:
Ah,

The alarm server configuration will not affect the PV alarm state, this alarm state is controlled by the values you have set in your record fields (HIHI ,LOLO, HHSV, LLSV...). 

The phoebus alarm server is built on top of epics.

The delay and other configuration only affect the behavior of the phoebus alarm server. The alarm server will not record this pv as being in alarm state until 7 seconds...so if you have the alarm tree/table app open, or you have automated actions, etc... the apps will not show this pv as being in alarm, or the automated actions will not be executed until 7 seconds have passed.
At NSLS2 this was one of the behaviors we have verified during our testing process.

MaticPogacnik, 2020-05-22:
Hi,

I do understand this, maybe my question/problem was not explained well, let me try again.

From the documentation: `Alarm Delay: Only alarm if the trigger PV remains in alarm for at least this time.`  https://github.com/ControlSystemStudio/phoebus/blob/master/app/alarm/ui/doc/index.rst

In the example above I have delay of 7 seconds, and the PV changes alarm state between the no alarm and alarm state many times in this time frame (purpose of the camonitor screen are timestamps not showing that it's in alarm). I was expecting that every time state from no alarm to alarm changes delay timer is started from 0. And in this example Alarm server should not show active alarm at all, but it does.

If this is not the case how does the delay timer reset? Should be PV in no alarm state for some time?

Was this logic maybe changed from 4.6.0 tag?

kasemir, 2020-05-22:
The basic alarm delay/count behavior is described in the online help, https://github.com/ControlSystemStudio/phoebus/blob/master/app/alarm/ui/doc/index.rst#behavior

For details on the filter, see unit test of the alarm server logic, https://github.com/ControlSystemStudio/phoebus/blob/master/services/alarm-server/src/test/java/org/phoebus/applications/alarm/server/AlarmLogicUnitTest.java, specifically the `test*Delayed*` tests.

Your specific case seems to match the one checked from https://github.com/ControlSystemStudio/phoebus/blob/master/services/alarm-server/src/test/java/org/phoebus/applications/alarm/server/AlarmLogicUnitTest.java#L559 on:

1) PV goes into MINOR alarm. This starts the 'delay' timer.
2) PV goes into MAJOR alarm.
3) PV back to MINOR alarm.
4) 'Delay' timer expires. PV is not OK, so we indicate & latch alarm. Specifically, the MAJOR alarm since that's the highest observed in the respective time period.

A simpler example where the delay suppresses short alarms (which should not happen in the first place, and having to use a delay on the alarm server side is a band aid, it would be better to fix this on the IOC side, but that's another story) is in `testDelayedButShort()`:

1) PV goes into MAJOR alarm. This starts the 'delay' timer.
2) PV recovers back to OK
3) 'Delay' timer expires. PV is OK, so we forget about the whole incident.

kasemir, 2020-05-22:
> I was expecting that every time state from no alarm to alarm changes delay timer is started from 0. 

No. The delay mechanism is specifically designed to catch PVs that go in/out/in/out/in/out of alarm all the time. We start the timer when going into alarm, we maximize the severity of observed alarms, and then check at the end:

a) Still in alarm -> Latch the highest observed alarm
b) Not in alarm -> Forget about it

For a noisy PV it is somewhat up to chance if you find it in or out of alarm at the end of the delay. As mentioned, noisy PVs should not happen in the first place. The 'count' can be used to catch those, like alarm if there are 5 or more triggers within 60 seconds.

No, the logic and the test should not have changed for about 10 years, the AlarmLogic and AlarmLogicUnitTest UnitTest have been imported from the Eclipse-based version.

kasemir, 2020-05-22:
You're right that the rather short description "Alarm Delay: Only alarm if the trigger PV remains in alarm for at least this time." is misleading. I'll update it.

kasemir, 2020-05-22:
Sorry, back up. Looks like I really need to document this and add one or more tests.
Again there's no change to the logic, all the decade old tests still work.
.. and the logic should behave as you expected, the delay _does_ reset when the PV returns to OK.
So your example PV which goes in and out of alarm reach second, with a delay of 7 won't actually be considered an alarm:

[image]

Note how the PV goes in out of alarm every second, but the alarm system ignores that because it doesn't last fo 7 seconds.
When I change the SCAN period to "2 second", the alarm lasts long enough to be considered.
Or, back at SCAN ".1 second", I can also set a count, like "3 (or more) triggers within 7 seconds":

[image]

Looks like there is a bug, though, when once setting an alarm count and then configuring the alarm count back to 0. The alarm server will still act as if an alarm count has been configured unless I either restart the alarm server, or remove and re-add the PV to the config. Need to look at that.

kasemir, 2020-05-22:
Is that what you're seeing, did you try various delay and alarm count settings, at one time the count was larger than zero, and when you configured it back for  count=0, it still latched, and that's what we see in your original screen shot?

MaticPogacnik, 2020-05-25:
Hi @kasemir ,

Thank you for  your time and looking into this. Will test this as soon as possible.

Best regards,
Matic

closed 2020-05-25 by MaticPogacnik