ControlSystemStudio/phoebus #1356: The core logbook update

shroffk, 2020-05-27:
The core logbook module needs to be updated
- improve javadocs
- add documentation of how to contribute
- review the api and determine potential reductions

shroffk, 2020-05-27:
In api review, making all the delete operations default in the client interface...

kasemir, 2020-05-27:
Is there a logbook system that allows end-users to delete entries?

shroffk, 2020-05-27:
Delete in olog is used equivalent to hide.
You can delete log entries in the sense that they are no longer visible.

Use case:
Users enter sensitive data into the log entry, some password, some procedure/action which should not be repeated...
The delete allows you to hide this log entry so that sensitive data can be protected or bad/dangerous practices do not show up in searches and are thus not repeated.

This is very uncommon and thus it seems unecessary to force all clients to implement it.

shroffk, 2020-05-27:
Actually on more thought I am making all other operations optional too.

At the heart of the logbook system you need.

1. to create a log entry
2. retrieve a log entry
3. search for a log entry

Except these all others are being made optional.

Operations like list logbook and list tags which are used by the UI will by default return empty lists.
Does this make sense @kasemir @georgweiss

shroffk, 2020-05-27:
This should really help simplify and clean up the client code.

kasemir, 2020-05-27:
For the SNS logbook, the delete operations could be removed. Similar to how you describe the OLog, 'delete' is only used in very rare cases, be performed by the administrator directly in the RDB, not by an end user via some GUI.
While it may be inconvenient to end users that you cannot delete or edit entries, not even fix simple typos, the point of the logbook is that it's a permanent record.

georgweiss, 2020-05-27:
About listing logbooks and tags: not sure why you ask, but this to me seems a key feature:
1) User visits logbook portal and will select a logbook.
2) User creates logbook entry and selects from tag list.

kasemir, 2020-05-27:
I think the point is yes, logbook and tag lists are useful, they are included in the UI, but there are defaults returning empty lists to simplify implementing the glue code for a specific logbook.
A proper logbook will end up implementing pretty much every method, but to get started, a "test" logbook could implement next to nothing and already compile/run.

georgweiss, 2020-05-27:
OK, now I understand the question. I do not have any opinion on the default behavior other than it should be easy to remove any "dummy" logbook when deploying to a production environment.

closed 2020-06-04 by shroffk