ControlSystemStudio/phoebus #1360: Question: global macros

georgweiss, 2020-05-28:
In RCP CS Studio the user could define macros on global level in the preferences editor. I cannot see a way to do the same in Phoebus...

pavel-ch, 2020-05-28:
For Display Builder, there is a preference in settings.ini:
[image]
Macros RES, PIC, PLT and OBJ are defined here.
Note $(hmi_project_location) is a Java property defined on the java command line, -D switch.

Not sure what if other tools use macros.

georgweiss, 2020-05-28:
OK, thanks. I was looking in the wrong place, apparently.

kasemir, 2020-05-28:
A good place to look for preference information is the help, https://control-system-studio.readthedocs.io/en/latest/preference_properties.html#display-builder-model:

```
# ----------------------------------------
# Package org.csstudio.display.builder.model
# ----------------------------------------

# Global macros, used for all displays.
...
macros=..
```

georgweiss, 2020-08-19:
Reopening to discuss a proposal:
Currently there are "built-in" macros like DID and DNAME (more...?). How about adding HOSTNAME (and maybe USERNAME)? Or add a way to fetch any system property through a macro, e.g. SYSTEM_PROPERTY:propertyName?

pavel-ch, 2020-08-19:
This works right now, you can use $(user.name) which comes from Java system properties. Use for any java property, see Help->About->Show Details->System Properties. Hostname is not directly there but can be included into Java properties via start script on java command line (-D switch).

georgweiss, 2020-08-19:
Thanks @pavel-ch . I'm aware of the command line option in the case of hostname, I just wanted to propose a more generic solution where hostname would be available as a built-in macro as that is not exposed as a native Java property like user.name. Our users are requesting it as they want to limit certain operations to only one control room machine at a time, and hence the hostname would be a suitable discriminator.

kasemir, 2020-08-19:
On Unix & Mac OS, `$(HOSTNAME)` is defined as an environment variable, and thus also available in macros.
If you depend on the hostname, best also define that on your windows system as an environment variable.
Getting the hostname from the code seems trivial but isn't.
The technical issue is that there's a difference between the `$(HOSTNAME)` that's somehow assigned to the computer, and the name associated with the IP address. For the former, you can use `$(HOSTNAME)` or run the `hostname` command when on Linux or Mac OS. On Windows, I'm not sure there is such a thing.
For the latter, you can call `InetAddress.getLocalHost().getHostName()` in Java on any host, but it can hang for a while as the library it connecting to DNS, trying to figure out who we are.
The IP address might not resolve into any name.
A name in DNS can have many aliases. These can only be resolved in one direction properly: name to address. The reverse direction is ambiguous. Which one is the "official" name?
A host can have many different IP addresses - and each address can have many different names.
Bottom line is that `$(HOSTNAME)` is already supported on Linux and Mac OS, and on Windows your best option might be to put something in the startup script to define an equivalent `%HOSTNAME%`.

georgweiss, 2020-08-20:
Thanks @kasemir .
FYI: the HOSTNAME environment variable trick does not work on my Macbook, running Mac OS 10.15.5, even though it of course is defined as a system environment variable.
You are of course right that getting it from code may not  provide useful results. And it incurs some overhead and will depend on the responsiveness of the DNS system and its data quality.
I will go for the command line option.

closed 2020-08-20 by georgweiss

kasemir, 2020-08-20:
Regarding `$HOSTNAME` on Mac, it's curious. Inside a Mac OS `Terminal`, I find it defined, as can be seen via `echo $HOSTNAME`.
But when starting Phoebus, it's not on the list of environment variables (Help, About, ...)

Turns out the variable is defined but not _exported_. When I `export HOSTNAME` before running phoebus, it's available inside phoebus.

Now I cannot guarantee that `$HOSTNAME` is defined (but not exported) on all Macs. Maybe it's only defined here because of some site-specific additions. Same actually for Linux. To guarantee that it's available, you can add
```
export HOSTNAME=`hostname`
```
to the `phoebus.sh` start script for Linux / Mac.
I've just added it to the Mac application wrapper that I use for the SNS build, https://github.com/kasemir/phoebus-sns/commit/dc02e6194666ce8db07f6a6aede8b6cf1556bf7a